cocotb/cocotb #873: running test_endian_swapper_hal under python 3

tachwali, 2019-03-18:
I have tried to run the test_endian_swapper_hal example under python 3.
I got the following error:
"ImportError: /home/****/source/cocotb/examples/endian_swapper/tests/../cosim/io_module.so: undefined symbol: Py_InitModule"

I then fixed it by modifying io.c as shown below:
```
#include "io.h"
#include <Python.h>
#include "io_module.h"

// Python read function just takes an offset in bytes and a buffer
static PyObject *pWrFunction;
// Python read function just takes an offset in bytes and returns a value
static PyObject *pRdFunction;

// Functions called by Python
static PyObject *set_write_function(PyObject *self, PyObject *args) {

    pWrFunction = PyTuple_GetItem(args, 0);
    Py_INCREF(pWrFunction);

    PyObject *retstr = Py_BuildValue("s", "OK!");
    return retstr;
}

static PyObject *set_read_function(PyObject *self, PyObject *args) {

    pRdFunction = PyTuple_GetItem(args, 0);
    Py_INCREF(pRdFunction);

    PyObject *retstr = Py_BuildValue("s", "OK!");
    return retstr;
}

// Functions called by C (exported in a shared library)
unsigned int IORD(unsigned int base, unsigned int address) {

    unsigned int value;

    if (!PyCallable_Check(pRdFunction)) {
        printf("Read function not callable...\n");
        return 0;
    }

    PyObject *call_args = PyTuple_New(1);
    PyObject *rv;

    PyTuple_SetItem(call_args, 0, PyInt_FromLong(base + address));

    rv = PyObject_CallObject(pRdFunction, call_args);
    value = PyInt_AsLong(rv);

    if (PyErr_Occurred())
        PyErr_Print();

    Py_DECREF(rv);
    Py_DECREF(call_args);
    return value;
}

int IOWR(unsigned int base, unsigned int address, unsigned int value)
{

    if (!PyCallable_Check(pWrFunction)) {
        printf("Write function isn't callable...\n");
        return -1;
    }

    PyObject *call_args = PyTuple_New(2);
    PyObject *rv;

    PyTuple_SetItem(call_args, 0, PyInt_FromLong(base + address));
    PyTuple_SetItem(call_args, 1, PyInt_FromLong(value));

    rv = PyObject_CallObject(pWrFunction, call_args);

    if (PyErr_Occurred())
        PyErr_Print();

    Py_DECREF(rv);
    Py_DECREF(call_args);

    return 0;
}

static struct PyModuleDef moduledef1 = {
    PyModuleDef_HEAD_INIT,
    "io_module",
    "",
    -1,
    io_module_methods    
};

PyMODINIT_FUNC
initio_module(void)
{
    PyObject* io_module;
#if PY_MAJOR_VERSION >= 3
   io_module = PyModule_Create(&moduledef1);
#else
    io_module = Py_InitModule("io_module", io_module_methods);
#endif
    if (!io_module) {
        printf("Failed to load io_module\n");
        exit(1);
    }
#if PY_MAJOR_VERSION >= 3
   return io_module;
#endif

}
```

But then I got the following error:
```
0.00ns CRITICAL cocotb.regression                         regression.py:135  in initialise                      Failed to import module test_endian_swapper_hal: dynamic module does not define module export function (PyInit_io_module)
    0.00ns INFO     cocotb.regression                         regression.py:136  in initialise                      MODULE variable was "test_endian_swapper_hal"
    0.00ns INFO     cocotb.regression                         regression.py:137  in initialise                      Traceback: 
    0.00ns INFO     cocotb.regression                         regression.py:138  in initialise                      Traceback (most recent call last):
                                                                                                                      File "/opt/anaconda3/lib/python3.6/site-packages/cocotb-1.1.0-py3.6.egg/cocotb/regression.py", line 133, in initialise
                                                                                                                        module = _my_import(module_name)
                                                                                                                      File "/opt/anaconda3/lib/python3.6/site-packages/cocotb-1.1.0-py3.6.egg/cocotb/regression.py", line 63, in _my_import
                                                                                                                        mod = __import__(name)
                                                                                                                      File "/home/ytachwali/source/cocotb/examples/endian_swapper/tests/test_endian_swapper_hal.py", line 36, in <module>
                                                                                                                        import io_module
                                                                                                                    ImportError: dynamic module does not define module export function (PyInit_io_module)
                                                                                                                    
Traceback (most recent call last):
  File "/opt/anaconda3/lib/python3.6/site-packages/cocotb-1.1.0-py3.6.egg/cocotb/__init__.py", line 167, in _initialise_testbench
    regression.initialise()
  File "/opt/anaconda3/lib/python3.6/site-packages/cocotb-1.1.0-py3.6.egg/cocotb/regression.py", line 133, in initialise
    module = _my_import(module_name)
  File "/opt/anaconda3/lib/python3.6/site-packages/cocotb-1.1.0-py3.6.egg/cocotb/regression.py", line 63, in _my_import
    mod = __import__(name)
  File "/home/ytachwali/source/cocotb/examples/endian_swapper/tests/test_endian_swapper_hal.py", line 36, in <module>
    import io_module
ImportError: dynamic module does not define module export function (PyInit_io_module)
Cocotb initialisation failed - exiting
    0.00ns ERROR    cocotb.scheduler                            __init__.py:190  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
VCD info: dumpfile waveform.vcd opened for output.
1000000000.00ns ERROR    cocotb.scheduler                            __init__.py:190  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
make[1]: Leaving directory '/home/ytachwali/source/cocotb/examples/endian_swapper/tests'
```

Any clue what is the issue here? It seems that the simulator is not built when trying to build the test_endian_swapper_hal module.

eric-wieser, 2019-03-19:
Python 2's `initio_module` is Python 3's `PyInit_io_module`. You need to conditionally change that name based on the python version,

tachwali, 2019-04-04:
Thank you for your help! This actually worked!
I made a pull request #895 to apply this fix in the repo

eric-wieser, 2020-01-21:
Closing - see also #1302 which fixed a different python3 issue.

closed 2020-01-21 by eric-wieser