cocotb/cocotb #875: GHDL unit test failing

themperek, 2019-03-18:
Test for GHDL are failing. See: https://travis-ci.org/potentialventures/cocotb/jobs/507809192 
Same test pass in Questa.

First step is to add `GHDL_ARGS ?= --ieee=synopsys` to https://github.com/potentialventures/cocotb/blob/master/tests/designs/viterbi_decoder_axi4s/Makefile

TC01, 2019-07-26:
This still seems to be broken (in current master) even after adding ```--ieee=synopsys```:

```
/home/bjr/Programming/git/cocotb/tests/test_cases/test_iteration_vhdl/../../../tests/designs/viterbi_decoder_axi4s/src/dec_viterbi.vhd:29:8:error: identifier "dec_viterbi" already used for a declaration
*libraries*:1:1:error: previous declaration: library "dec_viterbi"
/home/bjr/Programming/git/cocotb/tests/test_cases/test_iteration_vhdl/../../../tests/designs/viterbi_decoder_axi4s/src/dec_viterbi.vhd:76:14:error: entity 'dec_viterbi' was not analysed
/usr/bin/ghdl:error: compilation error
```

I'm not familiar enough with ghdl/vhdl to know how to fix this, unfortunately.

cogsy23, 2019-07-27:
Just pushed PR #1052 which lets test_iteration_vhdl run.  However it still fails due to #1003.

garmin-mjames, 2020-01-03:
The travis build for GHDL skips a lot of tests that could be run because `TOPLEVEL_LANG` is not set, so it defaults to `verilog`.

Running `make test SIM=GHDL TOPLEVEL_LANG=vhdl` locally, there are some test failures, which might be expected.

Also, `test_array` causes a fatal error:
```
     0.01ns INFO     cocotb.regression                         regression.py:343  in execute                         Running test 5/5: test_read_write
     0.01ns INFO     ..st.test_read_write.0x7f9ed828a048       decorators.py:251  in _advance                        Starting test: "test_read_write"
                                                                                                                     Description: Test handle inheritance
     0.01ns DEBUG    cocotb.gpi                                 GpiCbHdl.cpp:128  in run_callback                    Generic run_callback done
     0.01ns DEBUG    cocotb.gpi                                 GpiCbHdl.cpp:126  in run_callback                    Generic run_callback
     0.01ns DEBUG    cocotb.gpi                                 GpiCbHdl.cpp:128  in run_callback                    Generic run_callback done
     0.01ns DEBUG    cocotb.gpi                                 GpiCbHdl.cpp:126  in run_callback                    Generic run_callback
     0.01ns DEBUG    cocotb.gpi                                 GpiCbHdl.cpp:128  in run_callback                    Generic run_callback done
     0.01ns DEBUG    cocotb.gpi                                 GpiCbHdl.cpp:126  in run_callback                    Generic run_callback
     0.01ns DEBUG    cocotb.gpi                                 GpiCbHdl.cpp:128  in run_callback                    Generic run_callback done
     0.01ns DEBUG    cocotb.gpi                                 GpiCbHdl.cpp:126  in run_callback                    Generic run_callback
     0.01ns INFO     cocotb.test                               test_array.py:52   in test_read_write                 Checking Generics/Parameters:
     0.01ns DEBUG    cocotb.gpi                                GpiCommon.cpp:256  in __gpi_get_handle_by_name        Searching for param_logic
     0.01ns DEBUG    cocotb.gpi                                GpiCommon.cpp:269  in __gpi_get_handle_by_name        Checking if param_logic is native through implementation VPI
     0.01ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:278  in initialise                      VPI: param_logic initialized with 1 elements
     0.01ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:219  in create_gpi_obj_from_handle      VPI: Created object with type was ???(41)
     0.01ns DEBUG    cocotb.gpi                                GpiCommon.cpp:279  in __gpi_get_handle_by_name        Found param_logic via VPI
     0.01ns DEBUG    cocotb.gpi                                GpiCommon.cpp:50   in check_and_store                 Checking array_module.param_logic exists
     0.01ns DEBUG    cocotb.param_logic                            handle.py:84   in __init__                        Created
/usr/local/bin/ghdl:internal error: verilog_wire_val
FATAL: We are calling up again
/mnt/d/git/cocotb_github/cocotb/share/makefiles/simulators/Makefile.ghdl:68: recipe for target 'results.xml' failed
make[1]: *** [results.xml] Error 1
make[1]: Leaving directory '/mnt/d/git/cocotb_github/tests/test_cases/test_array'
/mnt/d/git/cocotb_github/makefiles/../cocotb/share/makefiles/Makefile.sim:86: recipe for target 'sim' failed
make: *** [sim] Error 2
```

eine, 2020-01-26:
@garmin-mjames, which version of WSL are you using? Is it WSL2? Last time we tried, GHDL would not work on WSL1: ghdl/ghdl#551.

cmarqu, 2020-04-29:
The same error still occurs with ghdl 0.37 and Ubuntu (seen with https://gitpod.io/#https://github.com/themperek/cocotb/tree/gitpod).

cmarqu, 2020-04-29:
We should add debugging options for CI: https://ghdl.readthedocs.io/en/latest/development/Debugging.html

cmarqu, 2020-04-30:
With a command line like ``env SIM_ARGS="--trace-signals --vpi-trace" make SIM=ghdl TOPLEVEL_LANG=vhdl`` I was able to get the output
```
[...]
     0.00ns DEBUG    Scheduling coroutine test_read_write
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 1ps
     0.00ns DEBUG    Scheduling with Value(<Timer of 1.00ps at 0x7f29b5ded880>)
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 1ps
     0.00ns INFO     Checking Generics/Parameters:
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 1ps
     0.00ns DEBUG    Searching for param_logic
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 1ps
     0.00ns DEBUG    Checking if param_logic is native through implementation VPI
 vpi_handle_by_name ("array_module.param_logic", 0000000000000000) = 0000562216A673A0
 vpi_get (vpiType, 0000562216A673A0) = vpiParameter
 vpi_get (vpiType, 0000562216A673A0) = vpiParameter
 vpi_get (vpiType, 0000562216A673A0) = vpiParameter
 vpi_get (vpiSize, 0000562216A673A0) = 1
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 1ps
     0.00ns DEBUG    VPI: param_logic initialized with 1 elements
 vpi_get_str (vpiType, 0000562216A673A0) = "???"
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 1ps
     0.00ns DEBUG    VPI: Created object with type was ???(41)
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 1ps
     0.00ns DEBUG    Found param_logic via VPI
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 1ps
     0.00ns DEBUG    Checking array_module.param_logic exists
 vpi_get_time (0000000000000000, {mtype=vpiSimTime}) = 1ps
     0.00ns DEBUG    Created
 vpi_get_value (0000562216A673A0, {format=BinStr}) = /usr/local/bin/ghdl:internal error: verilog_wire_val
 vpi call callback 0000562216A694E0 cbEndOfSimulation
FATAL: We are calling up again
make[1]: *** [/workspace/cocotb/cocotb/share/makefiles/simulators/Makefile.ghdl:76: results.xml] Error 1
```
The ``???`` seems to come from https://github.com/ghdl/ghdl/blob/master/src/grt/grt-vpi.adb#L716

garmin-mjames, 2020-04-30:
Looks like the issue is here:
https://github.com/ghdl/ghdl/blob/2318615ede87be1f27e8d776d5fc30b814d8e505/src/grt/grt-vcd.adb#L482

cmarqu, 2020-04-30:
Some related discussion in https://gitter.im/ghdl1/Lobby?at=5eab34c914b48f0698a03ea0

cmarqu, 2020-05-04:
Oops, this was closed accidentally.

garmin-mjames, 2020-05-13:
xref #1770 
Partial fix was recently merged to GHDL.

themperek, 2021-02-10:
Currently failing test:
```
2021-02-10T06:32:34.8691049Z Failure in testsuite: 'all' classname: 'issue_330' testcase: 'issue_330_direct' with parameters 'all'
2021-02-10T06:32:34.8692347Z Failure in testsuite: 'all' classname: 'issue_330' testcase: 'issue_330_iteration' with parameters 'all'
2021-02-10T06:32:34.8693828Z Failure in testsuite: 'all' classname: 'test_deprecated' testcase: 'test_unicode_handle_assignment_deprecated' with parameters 'all'
2021-02-10T06:32:34.8695342Z Failure in testsuite: 'all' classname: 'test_timing_triggers' testcase: 'test_time_units_eq_None' with parameters 'all'
2021-02-10T06:32:34.8696965Z Failure in testsuite: 'all' classname: 'test_scheduler' testcase: 'test_last_scheduled_write_wins' with parameters 'all'
2021-02-10T06:32:34.8698450Z Failure in testsuite: 'all' classname: 'test_handle' testcase: 'test_string_handle_takes_bytes' with parameters 'all'
2021-02-10T06:32:34.8699748Z Failure in testsuite: 'all' classname: 'test_handle' testcase: 'test_integer' with parameters 'all'
2021-02-10T06:32:34.8701046Z Failure in testsuite: 'all' classname: 'test_handle' testcase: 'test_integer_overflow' with parameters 'all'
2021-02-10T06:32:34.8702365Z Failure in testsuite: 'all' classname: 'test_handle' testcase: 'test_real_assign_double' with parameters 'all'
2021-02-10T06:32:34.8703670Z Failure in testsuite: 'all' classname: 'test_handle' testcase: 'test_real_assign_int' with parameters 'all'
2021-02-10T06:32:34.8704965Z Failure in testsuite: 'all' classname: 'test_vhdl_access' testcase: 'check_enum_object' with parameters 'all'
2021-02-10T06:32:34.8706422Z Failure in testsuite: 'all' classname: 'test_vhdl_access' testcase: 'check_objects' with parameters 'all'
2021-02-10T06:32:34.8708123Z Failure in testsuite: 'all' classname: 'test_discovery' testcase: 'pseudo_region_access' with parameters 'all'
2021-02-10T06:32:34.8709475Z Failure in testsuite: 'all' classname: 'test_discovery' testcase: 'access_integer' with parameters 'all'
2021-02-10T06:32:34.8710780Z Failure in testsuite: 'all' classname: 'test_discovery' testcase: 'access_string_vhdl' with parameters 'all'
2021-02-10T06:32:34.8712075Z Failure in testsuite: 'all' classname: 'test_discovery' testcase: 'custom_type' with parameters 'all'
2021-02-10T06:32:34.8713319Z Failure in testsuite: 'all' classname: 'test_vhdl_block' testcase: 'block_iter' with parameters 'all'
2021-02-10T06:32:34.8714637Z Failure in testsuite: 'all' classname: 'test_array_simple' testcase: 'test_1dim_array_handles' with parameters 'all'
2021-02-10T06:32:34.8715993Z Failure in testsuite: 'all' classname: 'test_array_simple' testcase: 'test_1dim_array_indexes' with parameters 'all'
2021-02-10T06:32:34.8717341Z Failure in testsuite: 'all' classname: 'test_array_simple' testcase: 'test_exceptions' with parameters 'all'
2021-02-10T06:32:34.8718663Z Failure in testsuite: 'all' classname: 'test_iteration' testcase: 'discovery_all' with parameters 'all'
2021-02-10T06:32:34.8719987Z Failure in testsuite: 'all' classname: 'test_iteration' testcase: 'test_n_dimension_array' with parameters 'all'
```
Should we exclude those so the CI pass?

imphil, 2023-10-02:
GHDL is enabled in CI (and has been for a while), let's call this issue (over)done.