cocotb/cocotb #879: Use Black to have consistent formatting?

themperek, 2019-03-20:
When most PRs are in should we run [Black](https://github.com/ambv/black) and continue with the consistent format?

imphil, 2019-07-10:
General note on reformatting style enforcers: Forcing developers info a certain style is fine with me if the tool works well. I've used yapf in other projects, and it actually changed its behavior between yapf minor releases and Python versions, so different versions forced different output. That's unacceptable to me; in this case, a "weaker" formatter like autopep8 would be preferred by me.

On Black: What's the benefit of this tool vs yapf (or other similar tools)?

themperek, 2019-07-10:
>On Black: What's the benefit of this tool vs yapf (or other similar tools)?

I am not really expert although looks like Black is very popular.

cmarqu, 2019-08-01:
I found this in https://news.ycombinator.com/item?id=17155048 

> - YAPF would at times not produce deterministic formatting (formatting the same file the second time with no changes in between would create a different formatting); Black treats this as a bug;

Also, there is https://github.com/asottile/blacken-docs which will be nice.

cmarqu, 2019-08-02:
blacken-docs needs https://github.com/asottile/blacken-docs/pull/29 to support the ``python3`` lexer name we are using in our ``code-block``s.

cmarqu, 2019-10-14:
Related: #6

eric-wieser, 2020-03-30:
A middle ground would be #1547

eric-wieser, 2020-03-30:
There's some existing discussion against doing this starting here: https://github.com/cocotb/cocotb/pull/1239#issuecomment-565969287

nevsan, 2020-03-30:
Our organization uses a tool we developed called [fourmat](https://github.com/4Catalyzer/fourmat) to combine `black`, `flake8`, and `isort`. It is stable enough to enforce in CI and pretty much everyone is happy with the appearance of the results. Regardless of using that tool or not, it is valuable to have a "one liner" to address all of the style issues and make it as easy as possible to "onboard" people onto using the tool (there are always growing pains, but worthwhile in the end IMO).

ktbarrett, 2021-08-25:
I'm going to bring this back up. There is a lot of poorly formatted code in the repo, the imports and their lack of order are particularly annoying. I have used a combination of `black` and `isort --profile black` to format projects. `black` autoformats code consistently; the formatting does not change version to version. With `black`, `flake8` is mostly pointless outside of checking for complexity, which we do not do.

marlonjames, 2022-02-08:
`black` is out of *beta* as of 22.1.0: https://black.readthedocs.io/en/stable/change_log.html

Their stability policy: https://black.readthedocs.io/en/stable/the_black_code_style/index.html#stability-policy