cocotb/cocotb #888: How to use cocotb with python3

Martoni, 2019-03-29:
On my distribution (debian) if I launch python with just his name it launch python2.7:
```
$ python
Python 2.7.9 (default, Sep 25 2018, 20:42:16) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```

To launch python3 I have to add the '3' at the end : 
```
$ python3
Python 3.4.2 (default, Feb  7 2019, 06:08:06) 
[GCC 4.9.2] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```

But to launch cocotb tests I don't launch python directly, I'm using the Makefile. And it's always python2.7 that be used.
What is the good way to force cocotb to use python3 instead of python (2.7) ?

themperek, 2019-03-29:
https://linuxconfig.org/how-to-change-from-default-to-alternative-python-version-on-debian-linux

closed 2019-03-29 by themperek

jahagirdar, 2020-04-22:
Is there a better way of doing this without a system wide change?

I can update-alternatives locally, but to do so on the shared regression farms would mean months of bureaucracy...

themperek, 2020-04-22:
if you install with `pip3` or `python3 -m pip` it will use correct version now (this has change and is in master branch). You can also check latest documentation update. You can also use miniconda (just unzip) to have local version.