cocotb/cocotb #889: Responding to an Avalon MM master

Bamban, 2019-04-01:
Hello,
I read the docs and looked at the examples thoroughly but it seems that I've seen nowhere any mention of an Avalon-MM slave model.
I have a DUT which speaks through 2 Avalon bus: read and write. The masters are in my DUT. For the talk with my master writer I assume I'll use an Avalon monitor. But for the master reader I'm a little bit confused. Do you know if there is any existing model I can use?
(If not I might consider contributing. This is a great project you have here! I'm migrating because I found Qsys+ modelsim too frustrating and I wanted to switch back to ghdl)

Martoni, 2019-04-01:
Hi Bamban,

As I understand, you should use AvalonMemory to test your avalon-MM master.
You have an example in [Tests directory](https://github.com/potentialventures/cocotb/blob/master/tests/test_cases/test_avalon/test_avalon.py#L56).

Declare a memory as a python dict : 
` self.memdict = {value: value for value in range(0x1000)}`
Then connect this 'memory' to your AvalonMM class : 
```
        self.avl32 = AvalonMemory(dut, "master", dut.clk,
                                  memory=self.memdict,
                                  readlatency_min=0,
                                  avl_properties=avlproperties)
```

Bamban, 2019-04-02:
Thanks for the reply.
OK, I'll give it a try with this pointer. After all it seems quite logical.
Thanks again!

Bamban, 2019-04-04:
Sorry for reopening :/
It works like a charm for the master reader ! But but but ... not for the writer. I decided to go for the Avalon Memory to respond to my master writer. So I wrote this:
```python
self.slave_writer = AvalonMemory(dut, "amio", dut.clk,
                                  avl_properties={})
```

Here is my interface on the VHDL side:
```vhdl
entity pipeline_wrapper is
	generic(
					 ADDR_W : natural := 32; -- address width
					 SYMB_W : natural := 8; -- symbol width
					 N_SYMB : natural := 4; -- number of symbols
					 CONF_W : natural := 8; -- configuration bus width
					 N_CONF : natural := 4; -- number of config words
					 BCNT_W : natural := 5; -- burscount width
					 BRST_S : natural := 4 -- burst size
				 );
	port(
				clk                : in std_logic;
				resetn             : in std_logic;

				-- input parameters
				mode               : in std_logic;
				pointer_din        : in std_logic_vector(ADDR_W-1 downto 0);
				n_input            : in std_logic_vector(ADDR_W-1 downto 0);
				pointer_dout       : in std_logic_vector(ADDR_W-1 downto 0);
				n_output           : in std_logic_vector(ADDR_W-1 downto 0);
				ivalid             : in std_logic;
				oready             : out std_logic;

				--outputs
				done               : out std_logic;
				ovalid             : out std_logic;
				iready             : in std_logic;

				-- ami input reader
				amii_readdata      : in std_logic_vector(N_SYMB*SYMB_W-1 downto 0);
				amii_readdatavalid : in std_logic;
				amii_waitrequest   : in std_logic;
				amii_address       : out std_logic_vector(ADDR_W-1 downto 0);
				amii_read          : out std_logic;
				amii_byteenable    : out std_logic_vector(N_SYMB*SYMB_W/8-1 downto 0);
				amii_burstcount    : out std_logic_vector(BCNT_W-1 downto 0);
				amii_enable        : out std_logic;
				amii_writeack      : in std_logic; -- this signal is purely ignored

				-- ami output writer
				amio_waitrequest   : in std_logic;
				amio_address       : out std_logic_vector(ADDR_W-1 downto 0);
				amio_write         : out std_logic;
				amio_writedata     : out std_logic_vector(N_SYMB*SYMB_W-1 downto 0);
				amio_byteenable    : out std_logic_vector(N_SYMB*SYMB_W/8-1 downto 0);
				amio_burstcount    : out std_logic_vector(BCNT_W-1 downto 0);
				amio_enable        : out std_logic;
				amio_writeack      : in std_logic -- this signal is purely ignored
			);
end pipeline_wrapper;
```

Here is the error I get from result.xml:
```xml
cat sim_build/results.xml 
<testsuites name="results">
  <testsuite name="all" package="all" tests="0">
    <property name="random_seed" value="1554362926" />
    <testcase classname="test_pipe" name="test1" ratio_time="50.297614769175865" sim_time_ns="13460.000000999999" time="267.60712337493896">
      <failure message="Test failed with random_seed=1554362926" stderr="" stdout="'Forked coroutine _respond raised exception Write to object amio_waitrequest was scheduled during a read-only sync phase.'" />
    </testcase>
  </testsuite>
</testsuites>
```
Note that I did a small check with pdb. The Avalon Memory has the `writeable` flag:
```python
ipdb> p vars(self)
{'log': <cocotb.log.SimLog object at 0x7fd476a5ef28>, '_pending': <cocotb.triggers.Event object at 0x7fd476a5e780>, '_sendQ': deque([]), '_thread': <cocotb.decorators.RunningCoroutine object at 0x7fd476a5ea20>, 'entity': HierarchyObject(pipeline_wrapper), 'clock': ModifiableObject(pipeline_wrapper.clk), 'bus': <cocotb.bus.Bus object at 0x7fd476a5ecf8>, 'name': 'amio', '_burstread': False, '_burstwrite': True, '_readable': False, '_writeable': True, '_width': 512, 'dataByteSize': 64, '_mem': {}, '_val': , '_readlatency_min': 1, '_readlatency_max': 1, '_responses': [], '_coro': <cocotb.decorators.RunningCoroutine object at 0x7fd476213630>}
```
Anything wrong ?

Bamban, 2019-04-04:
I don't know what happened now, the error is gone ... I haven't changed my test code since this morning when posting (or so I think).
....
10 minutes after, the error is back ... and gone right away the next simulation! Weird weird ...

garmin-mjames, 2019-04-22:
That error is caused by something in your cocotb yielding on a ReadOnly trigger, and then the write happening after that trigger fires, before the next time step.

Bamban, 2019-04-23:
OK, I see. I'll try remove that ReadOnly. Thanks!

closed 2019-04-23 by Bamban