cocotb/cocotb #890: List all Signals in Design

yrrapt, 2019-04-01:
Hi,

I'm trying to figure out a way to get a list of all the signal nodes in the design during a test.

This would be helpful because I'm designing for space application where radiation can cause random bit flips.  I want to be able to randomly select a signal in the design and flip a bit to determine how well the system responds.

Thanks in advance for any help you could provide.

pwiecha, 2019-04-01:
https://github.com/lukedarnell/cocotb/blob/master/tests/test_cases/orconf2018/cocotb_orconf2018.pdf
Slide 33 may be helpful

yrrapt, 2019-04-03:
That looks like what I need, thanks for the help.

closed 2019-04-03 by yrrapt

cmarqu, 2019-04-03:
@yrrapt Interesting - see https://github.com/potentialventures/cocotb/issues/657#issuecomment-436408411 too. If you come up with something that you can share (not necessarily code, also experiences), please let us know.