cocotb/cocotb #892: TestSuccess interpreted as an error in forked coroutine

andresdemski, 2019-04-04:
Hello,
I update cocotb to 1.1 and i have some issues with error tracebacks and an unexpected behavioral of TestSuccess.  
Verilog:  
```verilog
`timescale 1ns/1ps
module toplevel;
endmodule
```
Test:  
```python
import cocotb
from cocotb.triggers import Timer
from cocotb.result import TestSuccess

@cocotb.coroutine
def raise_test_success():
    yield Timer(1, units='ns')
    raise TestSuccess("TestSuccess")

@cocotb.test()
def error_test(dut):
    cocotb.fork(raise_test_success())
    yield Timer(10, units='ns')
```
Output:
```
     0.00ns INFO     Running on Icarus Verilog version 10.2 (stable)
     0.00ns INFO     Python interpreter initialised and cocotb loaded!
     0.00ns INFO     Running tests with Cocotb v1.1.0 from /tmp/cocotb_prueba/venv/lib/python3.7/site-packages
     0.00ns INFO     Seeding Python random module with 1554336180
     0.00ns INFO     Found test cocotest.error_test
     0.00ns INFO     Running test 1/1: error_test
     0.00ns INFO     Starting test: "error_test"
                     Description: None
     1.00ns ERROR    Test Failed: error_test (result was TestError)
     1.00ns ERROR    Failed 1 out of 1 tests (0 skipped)
     1.00ns INFO     *****************************************************************************
                     ** TEST                 PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
                     *****************************************************************************
                     ** cocotest.error_test    FAIL            1.00          0.00      2406.02  **
                     *****************************************************************************
                     
     1.00ns INFO     *************************************************************************************
                     **                                 ERRORS : 1                                      **
                     *************************************************************************************
                     **                               SIM TIME : 1.00 NS                                **
                     **                              REAL TIME : 0.01 S                                 **
                     **                        SIM / REAL TIME : 186.92 NS/S                            **
                     *************************************************************************************
```

I have tests running in previous version and raising TestSuccess works as it's expected.

andresdemski, 2019-04-04:
This error happens in python3.6 and python3.7.
In python3.6 i got the following traceback:
```
     0.00ns INFO     Running on Icarus Verilog version 10.1 (stable)
     0.00ns INFO     Python interpreter initialised and cocotb loaded!
     0.00ns INFO     Running tests with Cocotb v1.1.0 from /usr/local/lib/python3.6/site-packages
     0.00ns INFO     Seeding Python random module with 1554382642
     0.00ns INFO     Found test cocotest.error_test
     0.00ns INFO     Running test 1/1: error_test
     0.00ns INFO     Starting test: "error_test"
                     Description: None
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/cocotb/decorators.py", line 283, in __call__
    return RunningCoroutine(self._func(*args, **kwargs), self)
  File "/tmp/cocotb_prueba/cocotest.py", line 7, in raise_test_success
    raise TestSuccess()
cocotb.result.TestSuccess
     0.00ns INFO     
     0.00ns ERROR    Test Failed: error_test (result was TestError)
     0.00ns ERROR    Failed 1 out of 1 tests (0 skipped)
```

eric-wieser, 2019-04-05:
I think this was a regression introduced in #640

commit 455c77bb (2019-05-09): Fixes #892
closed 2019-05-09 by imphil