cocotb/cocotb #893: No error traceback in forked coroutines

andresdemski, 2019-04-04:
Hello,
I don't know if it is an expected behavioral but when i have an error in a forked coroutine, i haven't got any traceback to know where is it.

For example:
```python
@cocotb.coroutine
def coroutine_with_undef():
    new_variable = undefined_variable
    yield Timer(1, units='ns')

@cocotb.test()
def error_test(dut):
    cocotb.fork(coroutine_with_undef())
    yield Timer(10, units='ns')
```

Output:
```
     0.00ns INFO     Running on Icarus Verilog version 10.2 (stable)
     0.00ns INFO     Python interpreter initialised and cocotb loaded!
     0.00ns INFO     Running tests with Cocotb v1.1.0 from /tmp/cocotb_prueba/venv/lib/python3.7/site-packages
     0.00ns INFO     Seeding Python random module with 1554336569
     0.00ns INFO     Found test cocotest.error_test
     0.00ns INFO     Running test 1/1: error_test
     0.00ns INFO     Starting test: "error_test"
                     Description: None
     0.00ns ERROR    Test Failed: error_test (result was TestError)
     0.00ns ERROR    Failed 1 out of 1 tests (0 skipped)
     0.00ns INFO     *****************************************************************************
                     ** TEST                 PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
                     *****************************************************************************
                     ** cocotest.error_test    FAIL            0.00          0.00         3.76  **
                     *****************************************************************************
                     
     0.00ns INFO     *************************************************************************************
                     **                                 ERRORS : 1                                      **
                     *************************************************************************************
                     **                               SIM TIME : 0.00 NS                                **
                     **                              REAL TIME : 0.00 S                                 **
                     **                        SIM / REAL TIME : 0.28 NS/S                              **
                     *************************************************************************************
                     
     0.00ns INFO     Shutting down...
```

andresdemski, 2019-04-04:
The tracebacks appears in python3.6 but not in 3.7

xnorme, 2019-04-05:
I’ve been also experiencing similar problem which makes debugging testbenches a real nightmare :( 
Python 3.7

kolojoe, 2019-04-18:
Yes, I just brought this up in the gitter channel.  Glad someone else has made an issue about this.  I think this is rather high priority and would like to see it moved up.

eric-wieser, 2019-04-19:
@andresdemski: can you give an example of a test that only gives a traceback in 3.7?

andresdemski, 2019-04-21:
the tracebacks aren't shown in 3.7.

imphil, 2019-04-28:
Updated title to better reflect what's going on.

ktbarrett, 2019-05-04:
This is due to the error handling behavior in `cocotb.decorators.coroutine.__call__`. It catches any errors that occur before the coroutine becomes a `RunningCoroutine`, uses `traceback.print_exc()` to print the stack trace and then throws a `TestError`. I think you'll find if you put a `yield NullTrigger()` before the erroring code the error reporting will start working.

I'm not certain why this causes an issue on 3.7 vs 3.6, but the behavior is bad. Exceptions are propagated through `yield`s now, so any throwing that occurs should simply be allowed to continue unravelling the stack. That try-except section should be removed.

imphil, 2019-05-04:
@ktbarrett thanks for your analysis. Would you mind contributing a patch in a pull request?

ktbarrett, 2019-05-05:
After looking closer at @andresdemski's issue, it is not the same as mine, and I cannot replicate my issue at home with iverilog. Never the less, I'm fairly certain I have a solution to his issue.

commit bb0dbc2b (2019-05-09): Potentially fixes #893
closed 2019-05-09 by imphil