cocotb/cocotb #901: system signals hanlding and raise TestFailure

Bamban, 2019-04-09:
Hello,
I would like to handle CTRL-C when some simulation is running. I try to make the simulation end normally.
 I used the code below:
```python
class pipeTB(object):

    def __init__(self, dut, debug=False):
        self.dut = dut
        self.clk = ""
        # defining handle
        signal.signal(signal.SIGINT, self.sigint_handle)
        ...

        
    def sigint_handle(self, signum, frame):
        self.set_sim_efficiency() # some stuff of mine, playing with ghdl waveopt
        self.dut._log.info("Simulator efficiency; " + str(self.sim_eff) + " cycles/s")
        # raise TestFailure("Caugh SIGINT, stopping simulation")
        # sys.exit(0)

```

When I hit CLTR-C during simulation, this is what I get when using `raise TestFailure`:
```python
ERROR: called callback function returned NULL
Failed to execute callback due to python exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/cocotb-1.1.0-py3.7.egg/cocotb/scheduler.py", line 336, in react
make: *** [/usr/local/lib/python3.7/dist-packages/cocotb-1.1.0-py3.7.egg/cocotb/share/makefiles/Makefile.sim:85: sim] Interrupt

    self._event_loop(trigger)
  File "/usr/local/lib/python3.7/dist-packages/cocotb-1.1.0-py3.7.egg/cocotb/scheduler.py", line 456, in _event_loop
    self.schedule(coro, trigger=trigger)
  File "/home/alban/repo/fanna/device/hw/cocotb_sim/test_pipe.py", line 119, in sigint_handle
    raise TestFailure("Caugh SIGINT, stopping simulation")
cocotb.result.TestFailure: Caugh SIGINT, stopping simulation
```
When I use sys.exit, I only have:
```python
ERROR: called callback function returned NULL
Failed to execute callback due to python exception
```

eric-wieser, 2019-04-10:
Can you give a complete test case?

Note that `sys.exit(0)` is just syntactic sugar for `raise SystemExit()`, which explains why the behavior is almost identical.

garmin-mjames, 2020-10-28:
There are signal handler callbacks for **VPI** (`cbSignal`) and **VHPI** (`vhpiCbSigInterrupt`).
I tend to doubt these are well-supported in simulators but may be worth investigating.