cocotb/cocotb #902: New User of Cocotb

morsingnet, 2019-04-10:
Hey Guys,
Im new to cocotb and wanted to try to the simple example, but I keep running into this issue:

$ git clone https://github.com/potentialventures/cocotb
Cloning into 'cocotb'...
$ cd cocotb/examples/endian_swapper/tests
$ make
make: cocotb-config: Command not found
Makefile:60: /Makefile.inc: No such file or directory
make: cocotb-config: Command not found
Makefile:61: /Makefile.sim: No such file or directory
make: *** No rule to make target '/Makefile.sim'.  Stop.

Anyone been there before?
Running on ubuntu 18.04

tmeissner, 2019-04-10:
Maybe you have to do `pip install ./cocotb` first, after checking out without as mentioned here:  https://github.com/potentialventures/cocotb#quickstart There was some work to support cocotb as package, but it seems not documented on the readthedocs site.

morsingnet, 2019-04-10:
Thanks @tmeissner, im moving into a bigger can-of-worms I think.

```
$ git clone https://github.com/potentialventures/cocotb
Cloning into 'cocotb'...
$ pip install ./cocotb
Processing ./cocotb
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ImportError: No module named setuptools
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-3WdWd2-build/
```
Im running this on a minimal ubuntu 18.04. I might have to install more packages?

morsingnet, 2019-04-10:
added 
`sudo apt-get install python-setuptools`
It works now :)
Thanks @tmeissner - im off to the testing land

closed 2019-04-10 by morsingnet

morsingnet, 2019-04-11:
BTW - the PIP install step is missing in this guide:

https://github.com/potentialventures/cocotb/blob/master/documentation/source/quickstart.rst#running-an-example

- Christian

themperek, 2019-04-11:
https://github.com/potentialventures/cocotb/pull/881