cocotb/cocotb #903: Assign std_logic_vectors via strings

kolojoe, 2019-04-18:
Would anyone be in favor of assigning vectors/std_logic via a string as opposed to first creating BinaryValue objects? This seems like a reasonable feature if cocotb is trying to emulate HDL syntax for assignments. The current `setimmediatevalue` function already checks for a few types (ints, floats, and ctypes structures) and creates BinaryValue objects out of them, so I don't feel like this feature would be out of place.

```python
my_pin <= 1    # Works
my_pin <= "H" # Doesn't work, but would like to see this work
my_pin <= BinaryValue("H") # Works
```

ktbarrett, 2019-05-05:
It would be trivial to add something to try to construct any value into `BinaryValue` at the beginning of the assignment operation. However, `BinaryValue` isn't the greatest type, there are optional but important parameters (binary representation and endianess) present on the constructor that shouldn't be assumed. Does `BinaryValue(X)` work as one would expect for every value of `X` that can be constructed successfully? If not, it's not something that should be done.

ktbarrett, 2024-05-18:
Closed by #3777.

closed 2025-02-17 by ktbarrett