cocotb/cocotb #904: test_external fails in Python 2

garmin-mjames, 2019-04-22:
When attempting to run test_function_raised_exception() in test_external.py, there is a syntax error when importing test test module:

```
     0.00ns CRITICAL cocotb.regression                         regression.py:135  in initialise                      Failed to import module test_external: 'return' with argument inside generator (test_external.py, line 347)
     0.00ns INFO     cocotb.regression                         regression.py:136  in initialise                      MODULE variable was "test_external"
     0.00ns INFO     cocotb.regression                         regression.py:137  in initialise                      Traceback:
     0.00ns INFO     cocotb.regression                         regression.py:138  in initialise                      Traceback (most recent call last):
                                                                                                                       File "...\cocotb\regression.py", line 133, in initialise
                                                                                                                         module = _my_import(module_name)
                                                                                                                       File "...\cocotb\regression.py", line 63, in _my_import
                                                                                                                         mod = __import__(name)
                                                                                                                     SyntaxError: 'return' with argument inside generator (test_external.py, line 347)

Traceback (most recent call last):
  File "...\cocotb\__init__.py", line 167, in _initialise_testbench
    regression_manager.initialise()
  File "...\cocotb\regression.py", line 133, in initialise
    module = _my_import(module_name)
  File "...\cocotb\regression.py", line 63, in _my_import
    mod = __import__(name)
  File "...\tests\test_cases\test_external\test_external.py", line 347
    yield
```

eric-wieser, 2019-04-22:
Why does CI not catch this?

garmin-mjames, 2019-04-22:
Also failing in Travis build but not reported as failing:
https://travis-ci.org/potentialventures/cocotb/jobs/522094580

eric-wieser, 2019-04-22:
Ah, I remember this - an error at import-time is not considered a failure...

imphil, 2019-04-28:
It's not Python2-only, I see this in Python 3 as well. (And yes, it's a shame CI doesn't catch this. I really hope to find time to fix that ...)

imphil, 2019-04-28:
I looked through the Travis logs, this is not a regression, the tests always failed with merging  https://github.com/potentialventures/cocotb/commit/15ee0dab3144deb23f6074ba9c713668acb6de2b (which added the tests in the first place)

closed 2019-05-05 by imphil