cocotb/cocotb #906: Code coverage in Modelsim

Jakub6, 2019-04-26:
Hello, 

does anybody have any idea, how to use Code Coverage in Modelsim (DE-64 2019.1)?

When I tried to use cocotb with modelsim and add the commands for code coverage then I got this warning 

Loading libvpi.dll # **Warning: (vsim-8634) Code was not compiled with coverage options.

Thanks.

tmeissner, 2019-04-26:
You have to activate code coverage when compiling your sources like this: `vcom test.vhd +cover=bcsf`

Jakub6, 2019-04-26:
thanks for prompt reply.

I did it and it works perfectly in case that design and tesbench is written in Verilog (test.v, test_tb.v). 

But in Cocotb, I use those commands:
     vlog -work work +cover:bcesxf -sv -timescale 1ns/100ps -mfcu +acc=rmb  test.v
     vsim -coverage -pli libvpi.dll work.test
and nothing happened, just the warning: Loading libvpi.dll # **Warning: (vsim-8634) Code was not compiled with coverage options.

closed 2019-04-30 by Jakub6

Jakub6, 2019-05-13:
SOLUTION:

It started to work after I changed the command for code coverage
<b>+cover:bcesxf</b> to  <b>-cover bcesxf</b>

<b>Resulting runsim.do</b>
vlib work
vmap -c
vmap work work
vlog -fsmverbose -work work +define+COCOTB_SIM -sv -timescale 1ns/100ps -mfcu +acc=rmb -cover bcesxf   test.v
vsim -c -togglenovlogints -coverage  -onfinish stop -pli libvpi.dll  work.test
run -all
coverage report -output REPORT.txt -srcfile=* -assert -directive -cvg -codeAll
coverage report -html -output covhtmlreport -annotate -details -assert -directive -cvg -code bcefst -verbose
quit

You can find example on my GitHub or follow the examples in Install Dir of Modelsim (<i>c:\MentorGraphics\examples\c_windows</i>)

tmeissner, 2019-05-13:
Strange. I've looked in a cocotb project of us and we use `+cover=bcfse` which works without problems. The only difference is that we use VHDL and Questa.