cocotb/cocotb #910: test_function_returns_exception fails with AssertionError in scheduler.py:335

imphil, 2019-04-28:
Looks like we're having bad luck with `test_external` and Travis not reporting issues.

We currently get the following assertion failure on Python 3 (only. The test cannot run in Python 2 as reported in #904)

~~~
 13.91ns INFO     cocotb.regression                         regression.py:319  in execute                         Running test 9/11: test_function_returns_exception
    13.91ns INFO     .._returns_exception.0x7f7acd9d56d8       decorators.py:239  in _advance                        Starting test: "test_function_returns_exception"
                                                                                                                     Description:  Test that exceptions can be returned by @function coroutines 
ERROR: called callback function returned NULL
Failed to execute callback due to python exception
Traceback (most recent call last):
  File "/home/travis/build/potentialventures/cocotb/cocotb/scheduler.py", line 335, in react
    assert not self._pending_triggers
AssertionError
~~~

This assertion triggers since it was first added in https://github.com/potentialventures/cocotb/commit/643c395d089b2c15666aaf74138c26e23c9d7383, but our test harness doesn't seem to recognize this (I'm looking into that.)

imphil, 2019-04-28:
@eric-wieser do you have an idea what's going on here?

eric-wieser, 2019-04-28:
I'll take a look. I know that I convinced myself that my scheduler changes were correct by adding asserts, so if they were actually firing all along there may be problems

eric-wieser, 2019-05-05:
Fix is thankfully trivial

closed 2019-05-05 by imphil