cocotb/cocotb #911: Test harness doesn't catch failing assertions

imphil, 2019-04-28:
In https://github.com/potentialventures/cocotb/issues/910 we have the problem that a test asserts, but our test harness still reports all tests as passing.

eric-wieser, 2019-05-06:
To be clear, the real problem here is that the simulator does not exit if a callback throws an exception.

In practice, callbacks are not "supposed to" throw exceptions, so the only way this can happen is if:

* We screw up (like #910)
* `KeyboardInterrupt` is thrown (#901)

All we really need to do in these cases is bail out fast. I don't think reporting per-test pass / fail is particularly necessary here/

ktbarrett, 2025-08-27:
Support for Python callbacks appropriately handling exceptions was added over several commits in the 2.0 development window (sorry I'm lazy).

closed 2025-08-27 by ktbarrett