cocotb/cocotb #914: some handle.py issues

ktbarrett, 2019-05-04:
A test I wrote failed, I've been unable to diagnose why at a higher-level, but it led me to look into handle.py where I found some rather [dubious code](https://github.com/potentialventures/cocotb/blob/63a06bf2b85e760bd2ea999555bb675861ef9a4d/cocotb/handle.py#L260). In the test I wrote, it was failing because `__getattr__` was recursing infinitely. The linked bit of code only functions if `self._sub_handles` is defined prior to the call to `__getattr__`, otherwise it will recurse infinitely. `self._sub_handles` is only defined when the `_discover_all` method is called. `_discover_all` is not deterministically called from what I can tell before `__getattr__` (or at least the discussed access).

There is also [this](https://github.com/potentialventures/cocotb/blob/63a06bf2b85e760bd2ea999555bb675861ef9a4d/cocotb/handle.py#L149), which throws an error I've run into a time or two. `object.__getattr__` does not exist in Python 3.

imphil, 2019-05-04:
Can you share the test (or a similar one) that caused the error?

ktbarrett, 2019-05-04:
I'll try to write a more focused test.

closed 2019-05-05 by ktbarrett