cocotb/cocotb #920: cocotb.hook does not do anything unless in a module present in COCOTB_HOOKS

eric-wieser, 2019-05-05:
Either we should document this, or we should just remove `cocotb.hook`. None of our tests test `cocotb.hook`. Is anyone using it?

imphil, 2019-05-05:
The hooks functionality was added in https://github.com/potentialventures/cocotb/pull/565 by @stuarthodgson and @chiggs. So it's most likely used or has been used by them. There's a tiny bit of documentation at https://github.com/potentialventures/cocotb/blob/b6f24c142b14ade52a99e8c8c3d2fd46d49a9e77/cocotb/__init__.py#L113-L114.

@stuarthodgson @chiggs 
- Are you using cocotb.hook() at the moment and is it still useful to you?
- Can you outline a quick use case?

stuarthodgson, 2019-05-05:
It is used by others are organisation so please don't go removing it. I'll try and dignup some documentation for it buts it's a bit esoteric.

imphil, 2019-05-05:
Thanks @stuarthodgson. A use case would be nice to make sure we can add tests to ensure it doesn't get broken.

stuarthodgson, 2019-05-07:
I'm backtracking on this. It can be killed with great predudice.

PR #2201 (2020-11-17): Deprecate cocotb.hook
Closes #920. Deprecates `cocotb.hook`.
PR #2200 (2020-11-22): Increase coverage in xunit_reporter.py
Adding more deprecations, removals, and tests to increase code coverage. While the deprecated code is still being counted towards the total coverage it will be removed after 1.5 releases and 2.0 development begins. If desired we can add `pragma: no cover` comments to deprecated code blocks to reflect post-removal coverage numbers.

~~Deprecate `cocotb.hook` Closes #920.~~ Remove internal dead code in xunit_reporter.py. ~~Deprecate `cocotb.utils.unpack` and `cocotb.utils.pack`.~~ Add test for a test failure that escapes the regression manager.
closed 2020-11-17 by ktbarrett