cocotb/cocotb #922: Improve exception handling in forked coroutines

imphil, 2019-05-06:
(Spin-off from #893.)

In the long run, i think we need a better approach to handling exceptions in forked coroutines. Right now the following doesn't work as intended:

```py
task = cocotb.fork(a_coroutine_that_fails_soon())
yield longer_than_soon()
try:
    yield task
except Exception as e:
    pass  # we never get as far as here right now, but I think we ought to
else:
    assert False
```

ktbarrett, 2019-05-06:
In the above example, the failure happens, and the outcome is stored. What happens if the forked coro is never yielded on? Where does the exception go?

eric-wieser, 2019-05-07:
> In the above example, the failure happens, and the outcome is stored.

That's what _should_ happen, but it's not what _does_ happen

ktbarrett, 2019-05-11:
I don't agree. `fork`ed coroutines throwing exceptions should causing `TestError`. Many uses of `fork` are meant to start up 'threads' (yes, not great terminology but it mirrors `fork`) that run forever and are never meant to be yielded upon. This is used in many monitors and drivers. If exceptions are captured and the test never yields on the drive/monitor 'threads' (if they are meant to run forever they won't be yielded), those exceptions are never seen. I can see the how the behavior is useful, but I think it should exist in a new interface.

ktbarrett, 2019-05-11:
One potential way to support both behaviors is to make `scheduler.add` work as listed in the OP, and make `fork` wrap that with a function that yields on the coroutine and throws a `TestError` if an exception is caught. This will maintain the old behavior of `fork` while allowing the new behavior in a new interface that simply aliases `scheduler.add` as `fork` does now.

Also,there would have to be a special case for `TestComplete`s otherwise the test would continue longer than desired.

eric-wieser, 2019-05-11:
> If exceptions are captured and the test never yields on the drive/monitor 'threads' (if they are meant to run forever they won't be yielded), those exceptions are never seen.

That doesn't have to be true. The test cleanup could collect exceptions from all the remaining "threads".