cocotb/cocotb #923: Release cocotb 1.2

imphil, 2019-05-06:
This bug is to prepare for a cocotb 1.2 release.

- [ ] Start "release process" guide to document what we're doing here for the next releases.
- [ ] Triage bugs and issues to find blockers. Mark them with Milestone 1.2.
- [ ] Investigate/document PyPi upload process for release candidate and release packages.
- [ ] Release notes
  - [ ] write release notes
  - [ ] Add them to the documentation (#775)
- [ ] Tag relase
- [ ] Mailinglist announcement

imphil, 2019-06-12:
@themperek I'll had a quick look through the issues and PRs and nothing is very urgent to hold off a release, so let's do it. Can you post here how to upload the package to PyPi, and I'm currently working on the release notes and we'll put it together tonight?

garzetti, 2019-06-12:
Maybe have a look at #949 and see if it should be included in cocotb 1.2

themperek, 2019-06-14:
@imphil Sorry for the delay. Should be as simple as creating tag after this is in https://github.com/potentialventures/cocotb/pull/952

Can try to fist with `VERSION=1.1.2rc1' (can be deleted later) **but one cannot re-upload the same version to pypi**

imphil, 2019-06-25:
Documentation is started here: https://github.com/potentialventures/cocotb/wiki/Release-Process

I'm now working on the release notes, ETA tomorrow.

cmarqu, 2019-08-15:
This can be closed now.

closed 2019-08-15 by themperek