cocotb/cocotb #928: Travis is failing on master

eric-wieser, 2019-05-07:
Seems to be caused by the merge of #917, even though the travis run on that commit passed.

eric-wieser, 2019-05-07:
With some more investigation - the "failing" tests are due to:
```
TestFailure('Failure from external source: This is a a fatal message that fails the test',)
```
while the passing tests are due to
```
SimFailure('Failing test at simulator request before test run completion: Simulator shutdown prematurely')
```

So there are two things wrong here:
* The test is bad, and should be checking for `TestFailure` (`expect_failure=True`) not `SimFailure` (`expect_error=True`)
* `SimFailure` vs `TestFailure` seems to vary by platform / python version (!)

imphil, 2019-05-07:
Thanks for looking into that @eric-wieser. I saw the failures and retriggered them before (as travis was having a really bad weekend with many tests sporadically failing for no good reason). But the current failures are sticking and we need to debug them before merging anything else.

I won't have time to look into that until tonight or tomorrow, so if you find anything please let me now (or create a PR).

eric-wieser, 2019-05-07:
One option would be just to mark the test as skipped, and come back to it later.

I was able to make the error go away on a local branch, although I still have no understanding of what makes the jobs different.

eric-wieser, 2019-05-07:
@imphil, can you restart https://travis-ci.org/potentialventures/cocotb/builds/528342843?

eric-wieser, 2019-05-07:
@imphil: Also, can you create a `ci-temp` branch pointing at 63a06bf2b85e760bd2ea999555bb675861ef9a4d? travis won't let me bisect without one

imphil, 2019-05-07:
> @imphil, can you restart / clone https://travis-ci.org/potentialventures/cocotb/builds/528342843?

done

> @imphil: Also, can you create a ci-temp branch pointing at 63a06bf? travis won't let me bisect without one

done as well.

eric-wieser, 2019-05-07:
Thanks, updated #931, #932, and #933 to use `ci-temp` as the base - hopefully that will reveal which other commit made #917 go from passing to failing

closed 2019-05-08 by eric-wieser