cocotb/cocotb #936: TestError and expect_error

ktbarrett, 2019-05-11:
The `cocotb.test` decorator can take an `expect_error` argument. Currently this is a boolean deciding that the given test passed if it got a `TestError`, and failing otherwise. There are many reasons why a test would fail, and usually only one that is actually expected. However, `TestError` muddles all errors, making `expect_error` work if *any* error is caught. Most good regression systems allow you to expect an error and specify what type of exception is anticipated, failing otherwise.

I'd like to see this
```python
class MyException(Exception): pass

@cocotb.coroutine
def throws_my_exception():
    raise MyException

@cocotb.coroutine
def throws_other_exception():
    raise NotImplementedError

@cocotb.test(expect_error=MyException)
def passing_test(dut):
    yield throws_my_exception()

@cocotb.test(expect_error=MyException)
def failing_test(dut):
    yield throws_other_exception()
```

This would require re-writing a good bit of the scheduler and regression manager to not use `TestError` and properly propagate exceptions. I can't see this being too big of a break, user code doesn't/shouldn't be catching or throwing `TestError`.

garmin-mjames, 2019-05-16:
I don't think it's safe to assume that non-core code wont use `TestError`.
But you could keep backwards compatibility if `expect_error=True` expects a `TestError` object.

ktbarrett, 2019-06-13:
@garmin-mjames 
You are correct about that. TestError can be used to short-circuit test failure in forked coroutines, a behavior not available in any other way. TestError isn't bad, but how it is being used in the regression manager/scheduler is an issue.

eric-wieser, 2019-09-18:
> This would require re-writing a good bit of the scheduler and regression manager to not use TestError and properly propagate exceptions. 

I think this might be done? Your proposed change looks very straightforward to me now.

commit 17ce4e51 (2019-11-22): Closes #936 users can specify an expected error type on tests
closed 2019-11-22 by imphil