cocotb/cocotb #938: make test fails

toddstrader, 2019-05-12:
I've followed the instructions in README.md and the endian_swapper test appears to run well with iverilog.  I also noticed that the Makefile at the top of the repo has a test rule, so I tried that out as well.  Doing so results in this after a while:

> make[1]: Leaving directory '/home/tstrader/git/cocotb/examples'
> ./bin/combine_results.py
> Failure in testsuite: 'all' classname: 'test_closedown' testcase: 'test_failure_from_system_task' with parameters 'all'
> Ran a total of 1 TestSuites and 144 TestCases
> Makefile:54: recipe for target 'test' failed
> make: *** [test] Error 1

Should I be able to run this out of the box?  Are there instructions on a developer setup which I am missing?  I'm starting to pull this apart to figure out what is going on, but if someone could point me in the right direction that would be appreciated.

themperek, 2019-05-12:
In principle yes.
It should look like this: https://travis-ci.org/potentialventures/cocotb/jobs/531310132

eric-wieser, 2019-05-12:
This looks like the bug I fixed very recently. What commit do you have?

toddstrader, 2019-05-13:
face <- palm
Thanks, that's totally it.  I was at c09c633030146d30f77b1d8cb0b80cc09c214b55 and now everything is passing after pulling from upstream.

On a related note, is there a way to test local changes other than pip installing every time?  I was just going to run cocotb-config out of the repo, but I see that it is not even there and must be constructed during the pip install.

closed 2019-05-13 by toddstrader

eric-wieser, 2019-05-13:
Use `pip install -e`, and the installed files will just be an indirection to your source for.