cocotb/cocotb #944: Ubuntu Python3 error

bellero2007, 2019-05-21:
Hello,
I am trying to use cocotb with python3.7 and ghdl 0.36 under ubuntu 19.04.
The simulation succeeds however it prints this error message:

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 481, in get_distribution
    dist = get_provider(dist)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 357, in get_provider
    return working_set.find(moduleOrReq) or require(str(moduleOrReq))[0]
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'cocotb' distribution was not found and is required by the application
```

For some reasons it appears that, even if installed through `pip3 install ./cocotb`, cocotb tries to use python2.7 for some tasks.

I am on cocotb 6919d9487959d079efb2a6a5f55f8b510627a662.

Any ideas?

[Here](https://github.com/potentialventures/cocotb/files/3202810/SampleProject.zip) you can find the project to reproduce the error.

Thanks

imphil, 2019-05-21:
You need to use either Python 2 or Python 3. If you want to use Python 3 (recommended), *either* run in an environment where the `python` binary name is Python 3 (e.g. in a venv, highly recommmended!), *or* set `PYTHON_BIN=python3` in your environment before running cocotb. Also install cocotb for the right Python version, i.e. use pip2 or pip3 as needed.

Please remove all build output before switching Python versions, otherwise you end up with a non-working combination of libraries.

bellero2007, 2019-05-21:
Now it works, thanks.
I did not find any reference in the documentation regarding the PYTHON_BIN variable, I suggest to insert it for the users of Debian-based distros, where `python` points to python2.7.

closed 2019-05-21 by bellero2007