cocotb/cocotb #945: Raising exception does not show traceback

ajeshp, 2019-05-25:
I am using cocotb1.2.dev1 installed via pip

When I raise an Exception in my monitor, I see that the test stops with a "Test Failed: <name_of_test> (result was TestError)", but no traceback is printed on the console and neither is the Exception message is printed.
The Exception I have raised is similar to AvalonProtocolError in the Avalon monitor.

Looking at the cocotb code, it looks like the problem is in the way the BaseException is handled in the_advance method in RunningCoroutine class. It sets the outcome and raises CoroutineComplete(), so the test is stopped properly, but nothing is logged to the console. 
Is this expected behavior?

ktbarrett, 2019-06-01:
This was fixed recently. bb0dbc2bc8cb6dbfadcc73fe0316b60f36b87b03. Using pip you can install a package directly from github.

ajeshp, 2019-06-01:
Thanks! I have installed cocotb after cloning the git repo,

closed 2019-06-01 by ajeshp