cocotb/cocotb #948: How to set individual bits of a bus signal?

killruana, 2019-06-09:
Take the following RTL

```
/**
 * @brief A 2-inputs multiplexer
 * @param i_data: Input data bus
 * @param o_data: Selected data
 * @param i_selector: Selector bus
 */
module Mux(i_data, o_data, i_selector);
    input [1:0] i_data;
    output o_data;
    input i_selector;

    assign o_data = i_data[i_selector];
endmodule
```

How I can set individuals bits of a signal in the testbench ?
I have tried the following code

```
import cocotb
from cocotb.result import TestError
from cocotb.triggers import Timer

@cocotb.test()
def test_mux(dut):
    dut.i_data[0] = 0b0
    dut.i_data[1] = 0b1

    dut.i_selector = 0b0
    yield Timer(10, 'ns')
    expected = 0b0
    if dut.o_data != expected:
        raise TestError("i_data={:02b}, i_selector={:01b}, expected {}, found {}".format(dut.i_data, dut.i_selector, expected, dut.o_data))

    dut.i_selector = 0b1
    yield Timer(10, 'ns')
    expected = 0b1
    if dut.o_data != expected:
        raise TestError("i_data={:02b}, i_selector={:01b}, expected {}, found {}".format(dut.i_data, dut.i_selector, expected, dut.o_data))
```

But the script fail with this error:

> IndexError: i_data(GPI_REGISTER) contains no object at index 0

killruana, 2019-06-09:
Never mind, I think I found the response in the logs.

> Sorry: Icarus does not currently support vpi_get_handle_by_index() for vpiNet objects (i_data).

cmarqu, 2019-06-09:
Have you tried setting the individual bits of a ``BinaryValue`` and assigning this to the vector? See e.g. https://cocotb.readthedocs.io/en/latest/_modules/cocotb/drivers/ad9361.html?highlight=binaryvalue

killruana, 2019-06-09:
Not yet. I was not aware of the existence of this class.

cmarqu, 2020-07-27:
See also #1952

closed 2020-11-25 by github-actions[bot]