cocotb/cocotb #957: Issue with multiple testcases and coroutines which fork more coroutines

jwprice100, 2019-06-20:
When forking co routines which fork co routines and running multiple test cases some kind issue occurs. The ultimate symptom is that the first test cases will pass, but the second test case will fail with an unhanded exception. This is true even if the two test cases contain identical code, suggesting some issue with cleanup between test cases. Running each test case individually works fine.  I've attached an example reproducing the issue. The attached test case also includes the simlog which includes a stack trace.

The ultimate cause of the exception seems to be ine 331 of the Scheduler:
assert not self._pending_triggers

Commenting out that line allows me to proceed, but I doubt that's the right answer.

[cocotb_test_case.zip](https://github.com/potentialventures/cocotb/files/3311461/cocotb_test_case.zip)

eric-wieser, 2019-06-21:
Thanks for the explanation - I'll take a stab at creating a minimal reproduction.

There are definitely issues with inter-test cleanup, and I think the eventual resolution will be to move all test-related cleanup out of the scheduler and into the regression manager

eric-wieser, 2019-06-21:
Tried to get this to work with `iverilog`, but obviously that doesn't work with vhdl. Converting it to a verilog file makes `RisingEdge(dut.O_TRIGGERS[i])` since it appears busses cannot be unpacked into their individual bits...

eric-wieser, 2019-06-21:
Ok, managed to reproduce on my machine with iverilog here: https://github.com/eric-wieser/cocotb/commit/f3ef9e3ef4d0ccb46d53b551d873481a7851adef

Will see how much I can remove and keep the bug...

eric-wieser, 2019-06-21:
Reduced to:
```python
import cocotb

from cocotb.triggers import Timer, RisingEdge

@cocotb.coroutine
def wait_edge(dut):
    yield [RisingEdge(dut.and_output)]

@cocotb.coroutine
def wait_nested(dut):
    cocotb.fork(wait_edge(dut))
    yield Timer(100)

@cocotb.coroutine
def setup(dut):
    """Utility coroutine to initialize a test case."""
    cocotb.fork(wait_nested(dut))

    yield Timer(1000)

@cocotb.test()
def test1(dut):
    yield setup(dut)

@cocotb.test()
def test2(dut):
    yield setup(dut)
```
```Makefile
include ../../designs/sample_module/Makefile

MODULE = issue_957
```

eric-wieser, 2019-06-21:
So, the actual description of this bug is

> Assertion fires if a `First` trigger has not yet fired at the point a test is cleaned up

What's happening here is centered around these lines:

https://github.com/potentialventures/cocotb/blob/9b68e181a37588dfed01690892c18420463dde64/cocotb/triggers.py#L555-L585

When the first test finishes, the following happens:
* `_wait` is removed from the scheduler
* The event loop stops, waiting for the next test to begin
* `_wait` is garbage-collected, now that nothing refers to it
* Line 575 throws `GeneratorExit`, which causes `_wait` to exit
* The local variables inside `_wait` are garbage collected, notably `waiters[0]`, which is waiting for the rising edge
* Line 566 throws `GeneratorExit`, _which is caught_
* Line 571 runs, and sets an event outside of the event loop!

jwprice100, 2019-06-24:
I tried the fixed from the pull request associated with #958 but I still see identical symptoms except the improved error handling provides

`# COUT:                                                                                                                        File "/home/domain/jprice/cocotb-master/cocotb-master/cocotb/scheduler.py", line 468, in unschedule
# COUT:                                                                                                                          coro.retval
# COUT:                                                                                                                        File "/home/domain/jprice/cocotb-master/cocotb-master/cocotb/decorators.py", line 117, in retval
# COUT:                                                                                                                          return self._outcome.get()
# COUT:                                                                                                                        File "/home/domain/jprice/cocotb-master/cocotb-master/cocotb/outcomes.py", line 54, in get
# COUT:                                                                                                                          raise self.error
# COUT:                                                                                                                        File "/home/domain/jprice/cocotb-master/cocotb-master/cocotb/decorators.py", line 141, in _advance
# COUT:                                                                                                                          return outcome.send(self._coro)
# COUT:                                                                                                                        File "/home/domain/jprice/cocotb-master/cocotb-master/cocotb/outcomes.py", line 37, in send
# COUT:                                                                                                                          return gen.send(self.value)
# COUT:                                                                                                                        File "/home/domain/jprice/cocotb_test_case/fpga/hdl/test.py", line 17, in clock_gen
# COUT:                                                                                                                          signal <= 0
# COUT:                                                                                                                        File "/home/domain/jprice/cocotb-master/cocotb-master/cocotb/handle.py", line 413, in __le__
# COUT:                                                                                                                          self.value = value
# COUT:                                                                                                                        File "/home/domain/jprice/cocotb-master/cocotb-master/cocotb/handle.py", line 140, in __setattr__
# COUT:                                                                                                                          return object.__setattr__(self, name, value)
# COUT:                                                                                                                        File "/home/domain/jprice/cocotb-master/cocotb-master/cocotb/handle.py", line 430, in <lambda>
# COUT:                                                                                                                          fset=lambda self,v: self._setcachedvalue(v),
# COUT:                                                                                                                        File "/home/domain/jprice/cocotb-master/cocotb-master/cocotb/handle.py", line 626, in _setcachedvalue
# COUT:                                                                                                                          cocotb.scheduler.save_write(self, value)
# COUT:                                                                                                                        File "/home/domain/jprice/cocotb-master/cocotb-master/cocotb/scheduler.py", line 486, in save_write
# COUT:                                                                                                                          self._writes_pending.set()
# COUT:                                                                                                                        File "/home/domain/jprice/cocotb-master/cocotb-master/cocotb/triggers.py", line 324, in set
# COUT:                                                                                                                          trigger()
# COUT:                                                                                                                        File "/home/domain/jprice/cocotb-master/cocotb-master/cocotb/triggers.py", line 296, in __call__
# COUT:                                                                                                                          self._callback(self)
# COUT:                                                                                                                        File "/home/domain/jprice/cocotb-master/cocotb-master/cocotb/scheduler.py", line 324, in react
# COUT:                                                                                                                          .format(self._pending_triggers)
[... cut ...] ler.InternalError: Expected all triggers to be handled but found [<cocotb.triggers._Event object at 0xe766e06c>]
# COUT:                                                                                                                      
# COUT:      3.50ns ERROR    cocotb.scheduler                              result.py:52   in raise_error                     Forked coroutine setup raised exception: Expected all triggers to be handled but found [<cocotb.triggers._Event object at 0xe766e06c>]
`

eric-wieser, 2019-06-26:
@imphil: Please reopen

imphil, 2019-06-26:
Looks like this was closed together with the merge.

ktbarrett, 2021-03-15:
Still broken as of 1.5. Testing shows that the issue is the re-entrant behavior of the scheduler.

An exception occurs in a `fork`'d coroutine which causes test end. This kills the test and unschedules everything from the scheduler. Then the scheduler returns, and the coroutine that called `fork` on the failing coroutine continues to run... We have discussed an "escape hatch" exception that can be thrown to unravel the stack of `_schedule()` calls without executing any more user code, but it's still not 100% foolproof. Users doing broad-spectrum exception catching around the fork may inadvertently catch and squash the escape hatch.

This should be address in 2.0 with a reimplementation/refactor of the scheduler.

jwprice100, 2021-03-15:
Forgive my ignorance, but couldn't the coroutine that the scheduler is running also be terminated? I imagine this has implications I don't appreciate it or you would have suggested it.

ktbarrett, 2021-03-15:
That coroutine is "terminated", it is unscheduled. However the user's code is still in the current stack, so it will be allowed to run again. If you want to see what I mean add `traceback.print_stack()` in `scheduler.cleanup()`. It's a very bad place to be. If instead of immediately calling a `_schedule()`, `fork` *scheduled* the task to run after the current one yields, you'll never end up in that situation.

ktbarrett, 2023-03-17:
I think this issue is resolved now. Use `cocotb.start_soon` or `cocotb.start` instead.

n2d7, 2023-03-20:
I still encounter this error. However, with 1.7.2, it throws a python stack overflow error instead of the assertion error in 1.6.2. 

```
26852.60ns INFO     cocotb.regression                  test1 passed
 26852.60ns INFO     cocotb.regression                  running test2 (2/2)
Fatal Python error: _Py_CheckRecursiveCall: Cannot recover from stack overflow.
Python runtime state: initialized

Current thread 0x00003c60 (most recent call first):
  File "E:\Test\cocotb_build\cocotb\handle.py", line 183 in __repr__
  File "E:\Test\cocotb_build\cocotb\triggers.py", line 399 in __repr__
  File "E:\Test\cocotb_build\cocotb\triggers.py", line 796 in <genexpr>
  File "E:\Test\cocotb_build\cocotb\triggers.py", line 795 in __repr__
  File "E:\Test\cocotb_build\cocotb\triggers.py", line 550 in __repr__
  File "E:\Test\cocotb_build\cocotb\scheduler.py", line 375 in _react
  File "E:\Test\cocotb_build\cocotb\triggers.py", line 678 in prime
  File "E:\Test\cocotb_build\cocotb\scheduler.py", line 632 in _resume_coro_upon
  File "E:\Test\cocotb_build\cocotb\scheduler.py", line 638 in _resume_coro_upon
  File "E:\Test\cocotb_build\cocotb\scheduler.py", line 638 in _resume_coro_upon
  File "E:\Test\cocotb_build\cocotb\scheduler.py", line 638 in _resume_coro_upon
  File "E:\Test\cocotb_build\cocotb\scheduler.py", line 638 in _resume_coro_upon
  File "E:\Test\cocotb_build\cocotb\scheduler.py", line 638 in _resume_coro_upon
  File "E:\Test\cocotb_build\cocotb\scheduler.py", line 638 in _resume_coro_upon
  File "E:\Test\cocotb_build\cocotb\scheduler.py", line 638 in _resume_coro_upon
  File "E:\Test\cocotb_build\cocotb\scheduler.py", line 638 in _resume_coro_upon
  File "E:\Test\cocotb_build\cocotb\scheduler.py", line 638 in _resume_coro_upon
  File "E:\Test\cocotb_build\cocotb\scheduler.py", line 638 in _resume_coro_upon
  File "E:\Test\cocotb_build\cocotb\scheduler.py", line 638 in _resume_coro_upon
  File "E:\Test\cocotb_build\cocotb\scheduler.py", line 638 in _resume_coro_upon
  File "E:\Test\cocotb_build\cocotb\scheduler.py", line 638 in _resume_coro_upon
  File "E:\Test\cocotb_build\cocotb\scheduler.py", line 638 in _resume_coro_upon
  File "E:\Test\cocotb_build\cocotb\scheduler.py", line 638 in _resume_coro_upon
.
.
.

```
It seems to be keep instantiating NullTriggers in `_resume_coro_upon `at the start of the test2. 
```
45047.60ns INFO     cocotb.regression                  test1 passed
 45047.60ns INFO     cocotb.regression                  running test2 (2/2)
 45047.60ns WARNING  cocotb.scheduler                   <NullTrigger for Start <Test adv_2_test> at 0x0000000013F91300>
 45047.60ns WARNING  cocotb.scheduler                   <NullTrigger for Trigger.prime() Error at 0x0000000013E0FC40>
 45047.60ns WARNING  cocotb.scheduler                   <NullTrigger for Trigger.prime() Error at 0x0000000013FF8800>
 45047.60ns WARNING  cocotb.scheduler                   <NullTrigger for Trigger.prime() Error at 0x0000000013FAA640>
 45047.60ns WARNING  cocotb.scheduler                   <NullTrigger for Trigger.prime() Error at 0x0000000013FB51C0>
 45047.60ns WARNING  cocotb.scheduler                   <NullTrigger for Trigger.prime() Error at 0x0000000013F9B4C0>
 45047.60ns WARNING  cocotb.scheduler                   <NullTrigger for Trigger.prime() Error at 0x0000000013FA5440>
 45047.60ns WARNING  cocotb.scheduler                   <NullTrigger for Trigger.prime() Error at 0x0000000013FB2980>
 45047.60ns WARNING  cocotb.scheduler                   <NullTrigger for Trigger.prime() Error at 0x0000000013FB2E00>
 45047.60ns WARNING  cocotb.scheduler                   <NullTrigger for Trigger.prime() Error at 0x0000000013F96480>
```