cocotb/cocotb #968: Make cocotb Triggers async-/await-able in Python 3.7

StefanD986, 2019-06-27:
It looks like Python 3.7 got stricter in the use of `async`/`await`, which breaks the usage of `@cocotb.coroutine` with `async`/`await`:

While the following still works in Python 3.6:
``` python
@coroutine
async def clock_gen(self, period=1000):
    while True:
        self.dut.clk <= 0
        await Timer(period / 2)
        self.dut.clk <= 1
        await Timer(period / 2)
```

It will fail in Python 3.7 with `TypeError: object RunningCoroutine can't be used in 'await' expression`.

Actually when using a Python 3.7 environment, then the IDE (PyCharm) already points out that it is not allowed: `Class 'Timer' does not define '__await__', so the 'await' operator cannot be used on its instances`

Since the cocotb [docs state](https://cocotb.readthedocs.io/en/latest/coroutines.html#async-functions) that the `yield` syntax is interchangeable with async/await this should be fixed. (Also `await` ... reads much nicer than `yield` and the `async` makes clear that this is an asynchronous function.)

Maybe related issues: #610

eric-wieser, 2019-06-28:
It sounds like you're just using an older version of cocotb that the one you're reading the docs for?

StefanD986, 2019-06-28:
I'm using cocotb@9b68e181a37588dfed01690892c18420463dde64 (dated June 16th), so a pretty recent version.

eric-wieser, 2019-06-28:
It sounds like you have different versions of cocotb installed in your 36 and 37 environments.

eric-wieser, 2019-06-29:
`Timer` inherits from `GPITrigger`, `GPITrigger` inherits from `Trigger`, and `Trigger` defines `__await__`:

https://github.com/potentialventures/cocotb/blob/d3ab38e86f88dcbe2d6b5cb737eeca29c25feca2/cocotb/triggers.py#L104

https://github.com/potentialventures/cocotb/blob/d3ab38e86f88dcbe2d6b5cb737eeca29c25feca2/cocotb/triggers.py#L132

https://github.com/potentialventures/cocotb/blob/d3ab38e86f88dcbe2d6b5cb737eeca29c25feca2/cocotb/triggers.py#L85-L91

I think something is outdated with your installation.

StefanD986, 2019-07-01:
Ah, I see its dynamically added based on the python version. That explains at least the IDE warning about `__await__` not being defined. (is that actually necessary? Pre-3.3 python should simply see the `__await__` as a normal method definition, shouldn't it? Didn' test that though.)

However independent of that: We were able to reproduce the bug also on a system with only Python 3.7 installed and cocotb from git master HEAD, so I think it is not related to #978.

Just to exclude any mess-up on our side, I will create a draft PR #987 that adds python 3.7 to the tox.ini to run the tests in python 3.7.

eric-wieser, 2019-07-01:
> is that actually necessary?

Yes, pre-3.3 `return x` is a `SyntaxError` inside generators

StefanD986, 2019-07-01:
Closing this for now, since the tests that are run under python 3.7 are passing. I will investigate more where the problem really is.

closed 2019-07-01 by StefanD986