cocotb/cocotb #973: Implicit declaration of function

andresdemski, 2019-06-27:
Current master branch (578293e17509410c0024356f431d03009032c813) is broken.

```console
$ make clean && make
....
cocotb_utils.c: In function ‘to_python’:
cocotb_utils.c:45:9: error: implicit declaration of function ‘exit’ [-Werror=implicit-function-declaration]
         exit(1);
         ^~~~
cocotb_utils.c:45:9: error: incompatible implicit declaration of built-in function ‘exit’ [-Werror]
cocotb_utils.c:45:9: note: include ‘<stdlib.h>’ or provide a declaration of ‘exit’
cocotb_utils.c:35:1:
+#include <stdlib.h>
 #else
cocotb_utils.c:45:9:
         exit(1);
         ^~~~
cocotb_utils.c: In function ‘to_simulator’:
cocotb_utils.c:54:9: error: incompatible implicit declaration of built-in function ‘exit’ [-Werror]
         exit(1);
         ^~~~
cocotb_utils.c:54:9: note: include ‘<stdlib.h>’ or provide a declaration of ‘exit’
cc1: all warnings being treated as errors
```

eric-wieser, 2019-06-27:
Introduced in #954, looks like it was merged without CI passing

imphil, 2019-06-27:
That was my fault, we recently had half-failing tests and I didn't double-check that the symptom is the same. It's now reverted.

closed 2019-06-27 by imphil