cocotb/cocotb #978: `Make` in virtualenv still links against system python

StefanD986, 2019-06-28:
## Description
I noticed that when `make`-ing cocotb, the virtual environment is not respected by parts of the makefiles. This results in `libsim.so` being linked to the wrong `libpython.so`

I have the following system:
 - Ubuntu 18.04
 - Python 3.6 installed system-wide (executable in `/usr/bin/`, libpython `/usr/lib/x86_64-linux-gnu/`)
 - Python 3.7 (alt)installed under `/usr/local/` with libpython installed under `/usr/local/lib/`
 - venv created from Python 3.7. 

I ~double~ triple checked that the paths are correct when activating the virtualenv (`python -c "import sys; print(sys.path)"` gives me the path with the virtualenv path in the front.
In fact even when I start a simulation it reports the correct python path in the log output:
`cocotb.gpi gpi_embed.c:103 in embed_init_python Using virtualenv at /home/stefan/.venvs/cocotb37/bin/python`

But when I enable debug logging, I can see that it actually uses a different python path:
`regression.py:131 in initialise Python Path [...],/usr/lib/python36.zip,/usr/lib/python3.6,/usr/lib/python3.6/lib-dynload`

So I dug deeper into the build of the cocotb native libraries. Checking the libraries that simulator.so is linked against, I can see that it is linked a gainst the Python 3.6 lib 🤨:
``` bash
ldd cocotb/build/libs/x86_64/libsim.so 
	[...]
	libpython3.6m.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0 (0x00007f519fc32000)
	[...]
```

Again, I triple checked that I ran the cocotb native libray build in the correct (py 3.7) virtual env. But somewhere in the makefiles it picks up the wrong path.

My suspicion is that maybe somewhere where a shell is spawned to get the python excutable and library paths, it does not get the environment variables passed.

## Steps to reproduce
 0. Start e.g. with Ubuntu 18.04
 1. Install a second python (e.g. python 3.7) that is different from the system python under `/usr/local/` prefix. (if building from source use --enable-shared in the configure step)
 2. Create a virtualenv with the secondary python.
 3. clone cocotb
 4. run `make` in cocotb folder
 5. check the libraries that simulator.so is linked against with `ldd cocotb/build/libs/x86_64/simulator.so`

### Expected results:
libsim.so should be linked against the libpython of the secondary python installation (in my case that would be the one in `/usr/local/lib/`

### Actual result:
libsim.so is linked against the default python under `/usr/lib/x86_64-linux-gnu/`

StefanD986, 2019-07-01:
The reason why this does not show up in the CI/CD builds seems to be that in the travis.yml `pyenv` is used to isolate the different python environments.

If I run the build & tests in a `pyenv` environment with python 3.7 as *preferred* python version, then  `libsim.so` is linked to the 3.7 `libpython.so`. 

If I run it in a (activated) virtualenv created with the builtin `venv` module it links to the wrong python 3.6 `libpython.so`

StefanD986, 2019-07-01:
@eric-wieser @imphil 
Found the root cause of the bug (plus potential fix ideas at the end):
https://github.com/potentialventures/cocotb/blob/cf24ef33ecbb9f1e7f25407c5ea4e4a923b9c680/cocotb/share/makefiles/Makefile.pylib.Linux#L60-L70

Its a tricky problem that is caused by several factors:
 - When two python versions with the same major version (e.g. 3.6 and 3.7) are installed on the system, then `python3-config` will point of course only to either `python3.6-config` or `python3.7-config`.
   - The Makefile tries to resolve the ambiguity by trying to call `$(PYTHON_BIN)$(PYTHON_VERSION)` where `$(PYTHON_VERSION)` is the full major and minor version number.

However...
 - The python executable(s) can [optionally have suffix(es?) appended to their names](https://www.python.org/dev/peps/pep-3149/), most commonly probably the `m` suffix, because its config option is enabled by default AFAIK. Not always is there a symlink from python3.7-config -> python3.7m-config (and all the other python executables or suffix versions)

### Long boring details for my case
So in my case I have as mentioned:
 - python 3.7
   - python executable(s): `/usr/local/bin/python3.7` and `/usr/local/bin/python3.7m`
   - python-config executable: `/usr/local/bin/python3.7m-config`
 - python 3.6
   - python executable(s): `/usr/bin/python3.6` and `/usr/bin/python3.6m` (with symlinks `/usr/bin/python3` and `/usr/bin/python3m`)
   - python-config executable: `/usr/bin/python3.6-config` and `/usr/bin/python3.6m-config` (which are actually symlinks to other locations) plus some symlinks... 

on my system installed.

If I run it in a python 3.7 venv, the makefile
 -  first tries `$(PYTHON_BIN)$(PYTHON_VERSION)-config`, which will be resolved to `python3.7-config`, which does not exist.
 - Next try in line is `$(PYTHON_BIN)$(PYTHON_MAJOR_VERSION)-config` which resolves to `python3-config` which unfortunatly links to `python3.6-config` in my case.

I am not using the virtualenv package, but instead the builtin venv package. But I am linking a maybe relevant issue in virtualenv anyway: https://github.com/pypa/virtualenv/issues/169

### (Maybe) fixes
use 
 - `distutils.sysconfig.get_config_var('LDVERSION')`
    - returns `3.7m` on my system
 - or `distutils.sysconfig.get_config_var('LIBRARY')` 
    - returns `libpython3.7m.a` on my system (note that this is only the static lib)

In the makefile.

Additionally it might be good to raise this issue with python.org, because I think it makes a lot of sense to also symlink the `python-config` in the virtualenv. Then we don't have to hunt via obscure internal variables for the right executable.

StefanD986, 2019-07-01:
And why this problem doesn't show up with `pyenv`(which is used in the CI/CD builds): `pyenv` ~symlinks to~ *executes a shell script* that forwards it to the `python-config` of the active python executable.

themperek, 2019-07-02:
Could you check if something like this would works for you?
```
PYLIBS:=$(shell $(PYTHON_BIN) -c 'from __future__ import print_function; from distutils import sysconfig; import os; print("-l"+os.path.splitext(sysconfig.get_config_var("LDLIBRARY"))[0][3:])')
```

StefanD986, 2019-07-03:
Yes that does the trick!

commit 79370727 (2019-07-09): Find PYLIBS more reliably

Read PYLIBS on macOS and Linux through Python, and not through python*-config to avoid mismatches between the Python version, caused by our rather fragile way of discovering python*-config.

Fixes #978
closed 2019-07-09 by imphil