cocotb/cocotb #979: dff example failing for vhdl

themperek, 2019-06-28:
[dff example](https://github.com/potentialventures/cocotb/tree/master/examples/dff) fails in questa and gvhdl. The Verilog version seems ok.

```
#      0.00ns INFO     cocotb.gpi                                  gpi_embed.c:277  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
#      0.00ns INFO     cocotb                                      __init__.py:131  in _initialise_testbench           Running tests with Cocotb v1.1.0 from Unknown
#      0.00ns INFO     cocotb                                      __init__.py:148  in _initialise_testbench           Seeding Python random module with 1561732811
#      0.00ns INFO     cocotb.regression                         regression.py:186  in initialise                      Found test dff_cocotb.run_test_001
#      0.00ns INFO     cocotb.regression                         regression.py:325  in execute                         Running test 1/1: run_test_001
#      0.00ns INFO     ...test.run_test_001.0x7fffea845b38       decorators.py:239  in _advance                        Starting test: "run_test_001"
#                                                                                                                      Description: Automatically generated test
#                                                                                                                      
#                                                                                                                      
#      0.00ns INFO     cocotb.scoreboard.dff                     scoreboard.py:217  in add_interface                   Created with reorder_depth 0
#   5000.00ns ERROR    cocotb.scoreboard.dff.output              scoreboard.py:142  in compare                         Received transaction differed from expected output
#   5000.00ns INFO     cocotb.scoreboard.dff.output              scoreboard.py:146  in compare                         Expected:
#                                                                                                                      1
#   5000.00ns INFO     cocotb.scoreboard.dff.output              scoreboard.py:150  in compare                         1
#   5000.00ns INFO     cocotb.scoreboard.dff.output              scoreboard.py:156  in compare                         Received:
#                                                                                                                      0
#   5000.00ns INFO     cocotb.scoreboard.dff.output              scoreboard.py:160  in compare                         0
#   5000.00ns WARNING  cocotb.scoreboard.dff.output              scoreboard.py:163  in compare                         Difference:
#                                                                                                                      0000      31                                               1
#                                                                                                                           0000 30                                               0
#                                                                                                                      
#   5001.00ns ERROR    cocotb.regression                         regression.py:304  in handle_result                   Test Failed: run_test_001 (result was TestFailure)
#   5001.00ns ERROR    cocotb.regression                         regression.py:206  in tear_down                       Failed 1 out of 1 tests (0 skipped)
#   5001.00ns INFO     cocotb.regression                         regression.py:379  in _log_test_summary               *********************************************************************************
#                                                                                                                      ** TEST                     PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
#                                                                                                                      *********************************************************************************
#                                                                                                                      ** dff_cocotb.run_test_001    FAIL         5001.00          0.01    522837.42  **
#                                                                                                                      *********************************************************************************
#                                                                                                                      
#   5001.00ns INFO     cocotb.regression                         regression.py:396  in _log_sim_summary                *************************************************************************************
#                                                                                                                      **                                 ERRORS : 1                                      **
#                                                                                                                      *************************************************************************************
#                                                                                                                      **                               SIM TIME : 5001.00 NS                             **
#                                                                                                                      **                              REAL TIME : 0.17 S                                 **
#                                                                                                                      **                        SIM / REAL TIME : 29822.88 NS/S                          **
#                                                                                                                      *************************************************************************************
```

themperek, 2019-07-18:
Also, bit older version of Questa fails for Verilog.

eric-wieser, 2019-07-18:
Can you try an old cocotb release?

themperek, 2019-07-18:
@eric-wieser Good idea.

It looks like it is this: https://github.com/cocotb/cocotb/commit/1b516dba0f7c0f40f3032afc7da8ccde2edb8c5c#diff-9b2c13ec4669330f843406138ea30117L128

@cmarqu

cmarqu, 2019-07-18:
Huh, what, our normal clock generator doesn't work for VHDL?

eric-wieser, 2019-07-18:
That commit is the last one that works, or the first one that fails?

themperek, 2019-07-18:
First to fail. (change how clock is generated)

cmarqu, 2019-07-18:
Ok, so the clocks seem to have different polarity now, https://github.com/cocotb/cocotb/blob/master/cocotb/clock.py#L100 starts with a 1, the original code with a 0. This can be fixed, maybe best by adding a new ``startvalue`` to ``Clock`` which will come handy for certain cases anyway (e.g. testpattern generation).
Still, it hints at a difference in scheduling if a Verilog toplevel works and VHDL doesn't.

cmarqu, 2019-07-18:
> Also a bit older version of Questa fails for Verilog.

With the same/similar difference in counts, or something else?

eric-wieser, 2019-07-19:
To be clear, the culprit is these lines?

https://github.com/cocotb/cocotb/commit/1b516dba0f7c0f40f3032afc7da8ccde2edb8c5c#diff-9b2c13ec4669330f843406138ea30117L115

Another change that snuck in there is that the clock became faster by a factor of 2

themperek, 2019-07-19:
For old Questa and Verilog:
```
#      -.--ns INFO     cocotb.gpi                                  gpi_embed.c:111  in embed_init_python               Did not detect Python virtual environment. Using system-wide Python interpreter.
#      -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
#      0.00ns INFO     cocotb                                      __init__.py:131  in _initialise_testbench           Running tests with Cocotb v1.1.0 from /faust/user/themperek/git/tmp/cocotb
#      0.00ns INFO     cocotb                                      __init__.py:148  in _initialise_testbench           Seeding Python random module with 1563516294
#      0.00ns INFO     cocotb.regression                         regression.py:188  in initialise                      Found test dff_cocotb.run_test_001
#      0.00ns INFO     cocotb.regression                         regression.py:322  in execute                         Running test 1/1: run_test_001
#      0.00ns INFO     ...test.run_test_001.0x7fffe9841048       decorators.py:255  in _advance                        Starting test: "run_test_001"
#                                                                                                                      Description: Automatically generated test
#                                                                                                                      
#                                                                                                                      
#      0.00ns INFO     cocotb.scoreboard.dff                     scoreboard.py:217  in add_interface                   Created with reorder_depth 0
#      5.00ns ERROR    cocotb.scheduler                              result.py:52   in raise_error                     Forked coroutine _monitor_recv raised exception: Unable to resolve to binary >z<
#                                                                                                                      Traceback (most recent call last):
#                                                                                                                        File "/faust/user/themperek/git/tmp/cocotb/cocotb/scheduler.py", line 705, in schedule
#                                                                                                                          result = coroutine._advance(send_outcome)
#                                                                                                                        File "/faust/user/themperek/git/tmp/cocotb/cocotb/decorators.py", line 166, in _advance
#                                                                                                                          raise CoroutineComplete()
#                                                                                                                      cocotb.decorators.CoroutineComplete
#                                                                                                                      
#                                                                                                                      During handling of the above exception, another exception occurred:
#                                                                                                                      
#                                                                                                                      Traceback (most recent call last):
#                                                                                                                        File "/faust/user/themperek/git/tmp/cocotb/cocotb/scheduler.py", line 496, in unschedule
#                                                                                                                          coro.retval
#                                                                                                                        File "/faust/user/themperek/git/tmp/cocotb/cocotb/decorators.py", line 132, in retval
#                                                                                                                          return self._outcome.get()
#                                                                                                                        File "/faust/user/themperek/git/tmp/cocotb/cocotb/outcomes.py", line 54, in get
#                                                                                                                          raise self.error
#                                                                                                                        File "/faust/user/themperek/git/tmp/cocotb/cocotb/decorators.py", line 156, in _advance
#                                                                                                                          return outcome.send(self._coro)
#                                                                                                                        File "/faust/user/themperek/git/tmp/cocotb/cocotb/outcomes.py", line 37, in send
#                                                                                                                          return gen.send(self.value)
[... cut ...]
#                                                                                                                      *************************************************************************************
#                                                                                                                      
#      5.00ns INFO     cocotb.regression                         regression.py:220  in tear_down                       Shutting down...
# ** Note: $finish
#    Time: 5001 ps  Iteration: 0  Instance: /dff
# End time: 08:04:54 on Jul 19,2019, Elapsed time: 0:00:00
# Errors: 0, Warnings: 0
```

For VHDL same as in first commnt.

themperek, 2019-07-19:
What is more interesting `ius`, `vcs` and `icarus` seem to be happy with any Verilog version.

cmarqu, 2019-07-19:
I'll do a PR fixing the clock polarity and revert the period change (though the latter should not make a difference), and try to find the real problem later.

cmarqu, 2019-07-19:
My fix is in https://github.com/cmarqu/cocotb/tree/doc-example-dff which depends on #1036 so that should go in first before I open this as a PR.

cmarqu, 2019-07-22:
BTW, my theory (without being able to look at waveform) about why it's behaving differently for VHDL and Verilog now is that ``RisingEdge`` and ``posedge`` work slightly differently - at time 0 in VHDL, we have a change from ``U`` to ``1`` which isn't a rising edge, but in Verilog, we start from ``0`` - and that _is_ a rising edge.

cmarqu, 2019-07-22:
> For old Questa and Verilog:

This seems to be something else entirely.

PR #1044 (2019-08-24): Extend docs. Restore clock behaviour before last change.
Closes #979.
closed 2019-08-24 by themperek