cocotb/cocotb #982: iverilog 11 (devel) fails tests/test_cases/issue_253

imphil, 2019-06-29:
**Originally reported in #253**

It looks like this issue is still present for me - I'm using Icarus Verilog on Linux Mint:

```
gfg@rom ~/git/cocotb $ uname -a
Linux rom 4.15.0-45-generic #48~16.04.1-Ubuntu SMP Tue Jan 29 18:03:48 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
gfg@rom ~/git/cocotb $ cat /etc/issue
Linux Mint 18.3 Sylvia \n \l
```

A simple `make` in the top level of the repo failed, because the `make` invocation in `tests/test_cases/issue_253` was failing as follows:

```
gfg@rom ~/git/cocotb/tests/test_cases/issue_253 $ make
Makefile:44: warning: overriding recipe for target 'results.xml'
/home/gfg/git/cocotb/cocotb/share/makefiles/simulators/Makefile.icarus:80: warning: ignoring old recipe for target 'results.xml'
make results.xml
make[1]: Entering directory '/home/gfg/git/cocotb/tests/test_cases/issue_253'
Makefile:44: warning: overriding recipe for target 'results.xml'
/home/gfg/git/cocotb/cocotb/share/makefiles/simulators/Makefile.icarus:80: warning: ignoring old recipe for target 'results.xml'
Skipping
make[1]: Leaving directory '/home/gfg/git/cocotb/tests/test_cases/issue_253'
PYTHONPATH=/home/gfg/git/cocotb/tests/test_cases/issue_253/build/libs/x86_64:/home/gfg/git/cocotb:/home/gfg/git/cocotb/tests/test_cases/issue_253: LD_LIBRARY_PATH=/home/gfg/git/cocotb/tests/test_cases/issue_253/build/libs/x86_64::/usr/lib:/usr/lib MODULE=issue_253 \
COCOTB_SIM=1 TESTCASE=issue_253_empty TOPLEVEL="" \
vvp -M /home/gfg/git/cocotb/tests/test_cases/issue_253/build/libs/x86_64 -m gpivpi sim_build/sim.vvp   
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:111  in embed_init_python               Did not detect Python virtual environment. Using system-wide Python interpreter.
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:276  in embed_sim_init                  Running on Icarus Verilog version 11.0 (devel)
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:277  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:131  in _initialise_testbench           Running tests with Cocotb v1.1.0 from /home/gfg/git/cocotb
     0.00ns INFO     cocotb                                      __init__.py:148  in _initialise_testbench           Seeding Python random module with 1561668890
Traceback (most recent call last):
  File "/home/gfg/git/cocotb/cocotb/__init__.py", line 168, in _initialise_testbench
    regression_manager.initialise()
  File "/home/gfg/git/cocotb/cocotb/regression.py", line 122, in initialise
    self._dut = cocotb.handle.SimHandle(handle) if handle else None
  File "/home/gfg/git/cocotb/cocotb/handle.py", line 778, in SimHandle
    raise TestError("Couldn't find a matching object for GPI type %d (path=%s)" % (t, path))
cocotb.result.TestError: Couldn't find a matching object for GPI type 0 (path=None)
Cocotb initialisation failed - exiting
     0.00ns ERROR    cocotb.scheduler                            __init__.py:191  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
VCD info: dumpfile waveform.vcd opened for output.
mkdir -p empty_top_level_result && mv results.xml empty_top_level_result/
mv: cannot stat 'results.xml': No such file or directory
Makefile:53: recipe for target 'empty_top_level' failed
make: *** [empty_top_level] Error 1
```

I don't know what simulator @leftink was using, but I'd guess from the mentions of FLI that it wasn't Icarus, which uses VPI. Digging a bit deeper it seems that `VpiImpl::get_root_handle()` was returning  a root handle of type `GPI_NONE`, which causes the crash down the line:

```
cocotb.result.TestError: Couldn't find a matching object for GPI type 0 (path=None)
```

"GPI type 0" is `GPI_NONE`. It seems that there are several top level handles available in Icarus, but not all of them are modules. Adding a check to `VpiImpl::get_root_handle()` to ensure it's returning an object handle of type `GPI_MODULE` (otherwise returning nothing) fixed this issue for me.

I will submit a pull request with my change in but my knowledge of VPI is very rusty, so I'm not sure if this is the correct long term solution. Feel free to reject it if there is a more appropriate solution.

_Originally posted by @FinnG in https://github.com/potentialventures/cocotb/issues/253#issuecomment-506521975_

imphil, 2019-06-29:
PR in https://github.com/potentialventures/cocotb/pull/974

imphil, 2019-06-29:
@FinnG could you bisect iverilog between `v10_1` and `master` to see when this change was introduced? I'm not sure if this is a bug on our side, or a bug on the iverilog side.

Besides, it would be great if the issue_253 test could be rewritten to support other simulators to actually cross-check the behavior of iverilog with ModelSim or VCS.

FinnG, 2019-06-30:
Thanks for explaining what's going on here @imphil. I ran a quick bisect as requested and found the following commit:

```
commit 988816c0b1350db9d045b4ac61b0e44bfdc98b8a (HEAD, refs/bisect/bad)
Author: Martin Whitaker <icarus@martin-whitaker.me.uk>
Date:   Sat Oct 21 15:04:25 2017 +0100

    Support separate compilation units in SystemVerilog.
    
    The compilation unit scope is now treated as a specialised form of
    package (with an automatically generated name). All items declared
    outside a design element are added to the current compilation unit
    package. Apart from when searching for a symbol, once we get into
    elaboration we can treat these just like any other package.
```

This is consistent with what I saw, which was a top level package with the name `$unit`. Unfortunatley, I may be of limited use rewriting test 253 as you suggested as I don't have easy access to ModelSim or VCS.

imphil, 2019-07-01:
Thanks @FinnG for the bisection! (Looks like icarus didn't have a release in quite some time.) The commit looks like this was intended, and your change is a good safety net anyways, so I merged that now.

Thanks!

commit cf24ef33 (2019-07-01): Make VpiImpl::get_root_handle() return a GPI_MODULE always (#974)

In the event that the user doesn't specify the `TOPLEVEL` environment
variable to indicate the DUT Cocotb is supposed to automatically
detect it by scanning which top level handles are available and
returning the correct one.

In Icarus, it seems there are multiple top levels, and not all of them
are of type `GPI_MODULE`. This was meaning an unexpected handle was
returned from `VpiImpl::get_root_handle()` causing a test to fail.

This change causes the VPI implementation to ignore all top level
handles that are not of type `GPI [... cut]
closed 2019-07-01 by imphil