cocotb/cocotb #986: gitter room points to the wrong repo

eric-wieser, 2019-07-01:
As discovered on gitter, the room is tied to the `OrionBytes/cocotb` repo (which does not exist).

The only way to fix this is for @chiggs or @OrionBytes to follow the instructions at https://gitlab.com/gitlab-org/gitter/webapp/blob/develop/docs/rooms.md#change-room-security-after-creation (or to add someone else as an admin, perhaps @imphil, and have them follow those instructions)

Related: gh-529

imphil, 2019-07-01:
I'm on it.

OrionBytes, 2019-09-25:
Hi guys, sorry - I was not looking here (and in Gitter at all) for a long time. What should I do to solve issues with the room?
I can't find settings that I should use for that. The same in the official gitter client, It seems that I'm not admin of this room anymore. Or maybe I'm missing something?
@imphil what should I do?

Thanks!

eric-wieser, 2019-09-25:
You need to talk to gitter support, there's no way to do it without their assistance

imphil, 2019-09-25:
Please write an email to support@gitter.im with roughly the following content. Please CC me (mail@philipp-wagner.com):

---

Hi,

Could you please make the following changes to the room https://gitter.im/cocotb/Lobby:

- Link the room to https://github.com/cocotb/cocotb
- Make the user imphil admin in this room.

Background: I created this room some time ago for the cocotb community, and never linked it to the correct repository. Since then, the official repository has become https://github.com/cocotb/cocotb, and imphil is maintainer of cocotb (see https://github.com/cocotb/cocotb/blob/master/CONTRIBUTING.md#maintainers).

Thank you!

OrionBytes, 2019-09-25:
I sent them email a little bit earlier today, so missed your proposition.
Here is info from support - 
>>
I have updated https://gitter.im/cocotb to be associated with the https://github.com/cocotb org tada. Anyone who is a member of that org, should be an admin.
If you still don't see admin permissions, run through these steps https://gitlab.com/gitlab-org/gitter/webapp/blob/develop/docs/faq.md#why-isnt-my-github-organisation-or-repos-appearing

eric-wieser, 2019-09-26:
Looks like it's been associated with the org. Thanks @OrionBytes!

@imphil, as follow-up: It might be valuable to associate the room with the repo and not the org, since then issue links will work in chat.

imphil, 2019-09-27:
It's all fixed now, enjoy!

imphil, 2019-09-27:
Thanks @OrionBytes for getting in touch with the Gitter support.

closed 2019-09-27 by imphil