cocotb/cocotb #991: Implement an assert report method

INTI-CMNB-FPGA, 2019-07-02:
I am new in CocoTB, and I didn't find something similar to the VHDL `assert ... report... severity`. I did a grep of assert in the CocoTB repository and I find that you are using the Python assert, which ends with an exception:
```
    0.00ns ERROR    Send raised exception:
                    Traceback (most recent call last):
                      File "/usr/local/lib/python2.7/dist-packages/cocotb/decorators.py", line 218, in _advance
                        return outcome.send(self._coro)
                      File "/usr/local/lib/python2.7/dist-packages/cocotb/outcomes.py", line 37, in send
                        return gen.send(self.value)
                      File "/home/ram/repos/core-comblock/testbench/comblock_tb.py", line 21, in test01_regs
                        assert 0
                    AssertionError
```
I am thinking about something that print an error message. I am using a trivial own function in my tests:
```
def do_assert(cond, msg):
    if not cond:
       raise TestFailure(msg)
```
Let me know if there is something similar in CocoTB. If no, what do you think? Maybe to add something like assert_failure, assert_error, assert_warning? (note?).

eric-wieser, 2019-07-03:
You can already do `assert cond, msg` if you just want a message to go with the failure

rodrigomelo9, 2019-07-03:
(it is my real user in GitHub)
I already know the Python assert... The problem that I see is that it finishes as an Uncaught Exception.
```
    0.00ns ERROR    Send raised exception:
                    Traceback (most recent call last):
                      File "/usr/local/lib/python2.7/dist-packages/cocotb/decorators.py", line 218, in _advance
                        return outcome.send(self._coro)
                      File "/usr/local/lib/python2.7/dist-packages/cocotb/outcomes.py", line 37, in send
                        return gen.send(self.value)
                      File "/home/ram/repos-git/core_comblock/testbench/comblock_tb.py", line 21, in test01_regs
                        assert 0, "Test"
                    AssertionError: Test
                    
    0.00ns ERROR    Test Failed: test01_regs (result was TestError)
```
I am talking about a method with finishes in a good way.

eric-wieser, 2019-07-03:
I plan to make cocotb handle assertions more gracefully in the imminent future - I have a local patch that makes a start on it.

rodrigomelo9, 2019-07-04:
What could be great is that when the simulation finds an unhandled exception, it ends aborting with an error other than 0 to evaluate its execution, without the need to parse results.xml

eric-wieser, 2019-07-04:
@rodrigo:

Sounds like a good idea. I think the way to go would be change the default makefile target to be one that fails if the tests fail. I don't think it would be valid for `make results.xml` to fail if some tests fail.

eric-wieser, 2020-05-07:
Closing in favor of #1774 - the `assert` behavior was already fixed in #1110. The complaint here is no longer about assertions at all, but about makefile exit codes.

closed 2020-05-07 by eric-wieser