cocotb/cocotb #994: GHDL can also use COMPILE_ARGS no ?

Martoni, 2019-07-03:
I saw in documentation that only VCS and Icarus can use COMPILE_ARGS. In cocotb/documentation/source/building.rst : 

>   ``COMPILE_ARGS``
>       Any arguments or flags to pass to the compile stage of the simulation. Only applies to simulators with a separate compilation stage (currently Icarus and VCS).

But it can also be used with GHDL has a saw in makefile.ghdl : 
```
# Compilation phase
analyse: $(VHDL_SOURCES) $(SIM_BUILD)
	cd $(SIM_BUILD) && \
	$(foreach SOURCES_VAR, $(filter VHDL_SOURCES_%, $(.VARIABLES)), \
		$(CMD) -a $(GHDL_ARGS) $(COMPILE_ARGS) --work=$(SOURCES_VAR:VHDL_SOURCES_%=%) $($(SOURCES_VAR)) && ) \
	$(CMD) -a $(GHDL_ARGS) $(COMPILE_ARGS) --work=$(RTL_LIBRARY) $(VHDL_SOURCES) && \
	$(CMD) -e $(GHDL_ARGS) $(COMPILE_ARGS) --work=$(RTL_LIBRARY) $(TOPLEVEL)

```

Maybe documentation should be updated no ?

imphil, 2019-07-03:
Thanks for reporting this. I'd appreciate a pull request for this issue. (Please do a quick grep check what other backends support this argument.)

Martoni, 2019-07-03:
Ok I'm doing it.

Martoni, 2019-07-03:
Done in pull request #995

closed 2019-07-03 by Martoni