cocotb/cocotb #1003: Support complex VHDL types with GHDL

cogsy23, 2019-07-06:
Nice work on this project so far, I'm glad to see it's still making progress.

I'm trying to simulate a VHDL entity with GHDL which includes some complex types in it's interface:

type Vector16 is record
  X, Y, Z : signed(15 downto 0);
end record;

type AxisEnum is (X, Y, Z);

However, these signals are all being mapped to the same ModifiableObject that std_logics are.  I can see there are children of ModifiableObject that should do what I want, but they aren't being used. I've been digging around (I have no experience with VPI) and found that the VPI layer is returning the same `vpiNet` type for all signals.

I'm not sure if the issue is in cocotb or GHDL, or the interface between the two...

cogsy23, 2019-07-06:
Actually, to clarify: the Enum is mapped to a ModifiableObject, the record doesn't appear in _sub_handles at all.

ktbarrett, 2019-10-16:
I don't understand. You are declaring a record type that has fields X, Y, and Z and then an enumerated type that can hold values X, Y, and Z. They are disjoint namespaces, they are not referencing the same objects, why should this work?

ktbarrett, 2021-01-31:
No additional info was added. Please reopen with clarification if you are still interested.

closed 2021-01-31 by ktbarrett