cocotb/cocotb #1004: Can't access verilog instance from vhdl top

mmdems, 2019-07-07:
Hi, I wrote simple modules and I can't access Verilog instance from VHDL top. I have mux4to1 (VHDL) and mux2to1 (Verilog). For example, I want to know d_o value of mux2to1 (with label mux2to1_2). 

Modules are as simple as possible, because my aim is just to know can I access verilog module from vhdl toplevel and vice versa.

When I try to access vhdl instance from vhdl toplevel the same way it works fine of course, but when I change language of inner module from vhdl to verilog I can't read the value.

What am I doing wrong? Or I didn't read documentation correctly? Is it possible to do what I want in general?

**mux4to1 (top level)**
```
entity mux4to1 is
    port(
        d_i  : in std_logic_vector(3 downto 0);
        sel  : in std_logic_vector(1 downto 0);
        d_o  : out std_logic
    );
end mux4to1;

architecture struct of mux4to1 is

    signal d_buf : std_logic_vector(1 downto 0);

    component mux2to1 is
        generic(
            TEST_GENERIC : integer := 2
        );
        port(
            d_i  : in std_logic_vector(1 downto 0);
            sel  : in std_logic;
            d_o  : out std_logic
        );
    end component;

begin

    mux2to1_1 : mux2to1 port map(d_i(3 downto 2), sel(1), d_buf(1));
    mux2to1_2 : mux2to1 port map(d_i(1 downto 0), sel(1), d_buf(0));
    mux2to1_3 : mux2to1 port map(d_buf, sel(0), d_o);

end struct;
```
**mux2to1**
```
module mux2to1(
    input   [1:0] d_i,
    input          sel,
    output         d_o
);

    assign d_o = sel ? d_i[1] : d_i[0];

endmodule
```
**Testbench**
```
import cocotb
from cocotb.triggers import Timer

@cocotb.test()
def test_mux(dut):
    """...txt..."""

    yield Timer(50)
    dut.d_i.value, dut.sel.value = 1011, 10
    yield Timer(50)

    dut.log.info(dut.TEST_GENERIC.value)
    dut.log.info(dut.mux2to1_2.d_o.value)
    dut.log.info(dut.d_o.value)
    yield Timer(40000)
```
**Makefile**
```
VERILOG_SOURCES=$(HOME)/cocotb_test/src/mux2to1.v
VHDL_SOURCES=$(HOME)/cocotb_test/src/mux4to1.vhd
COCOTB=$(COCOTB_SHARE_DIR)
TOPLEVEL=mux4to1
MODULE=mux4to1_tb
SIM=questa
GUI=0
TOPLEVEL_LANG=vhdl
SIM_ARGS= -voptargs="+acc" -t 1fs
include $(COCOTB)/makefiles/Makefile.inc
include $(COCOTB)/makefiles/Makefile.sim
```
**log:**
it's huge to paste here, so I share it here:
https://justpaste.it/2u13z

elgorwi, 2019-07-07:
Have you tried running the "mean" example? This example instantiates a VHDL module in a verilog top level. Have a look in the makefile, it uses the questa -mixedsvvh compile flag to enable mixed language simulation

mmdems, 2019-07-07:
Basically I have not any issues with compilation and simulation of this. I tried to compile and simulate it via Questa manually to ensure that everything is correct.

Trouble comes when I try to get the signal value of Verilog instance in my .py file. 

Also if I comment that one line and run it with “make” command everything (both compilation and simulation) works good.

```
@cocotb.test()
def test_mux(dut):
    """...txt..."""

    yield Timer(50)
    dut.d_i.value, dut.sel.value = 1011, 10
    yield Timer(50)

    dut.log.info(dut.TEST_GENERIC.value)
 # dut.log.info(dut.mux2to1_2.d_o.value)
    dut.log.info(dut.d_o.value)
    yield Timer(40000)
```

Will try “mean” example tomorrow (my bad, hmm, I was looking for mixed language examples but missed it)

cmarqu, 2019-07-08:
This part of the log looks fishy:
```
** Fatal: (vsim-4) ****** Memory allocation failure. *****
Attempting to allocate 140737432827944 bytes
Please check your system for available memory and swap space.
```
This number is close to 2^47, which may mean something, though I don't know what.

mmdems, 2019-07-09:
I run "mean" example. It works good and I successfully read the value of the inner instance and values of the wrapper

But I still don't know what's wrong with my multiplexers

cmarqu, 2019-07-09:
Does your mux work with a syntax like
```
dut.d_i <= 0b11
dut.sel <= 0
```
?

mmdems, 2019-07-09:
No, same error message

By the way, if I try to get the d_buf value (which is declared in VHDL toplevel and connected to d_o of Verilog instances) then I can get it correctly and I see no errors

cmarqu, 2019-07-09:
I now see I used the wrong widths above, taking them from the submodules... maybe there is also some internal confusion about the signals with the same name but on different hierarchy levels/different language domains.
You could try making the port names unique and see if that makes a difference.

mmdems, 2019-07-09:
> You could try making the port names unique and see if that makes a difference.

Nah, brings no result

cmarqu, 2019-07-09:
Does Questa have options to be more verbose about FLI problems?
Also, running cocotb with the environment variable ``COCOTB_LOG_LEVEL=DEBUG`` may report something interesting.

TrampHRC, 2019-07-12:
Hi,
I just solved a similar problem thanks to you. my problem:
- Questasim 10.6c
- cocotb 1.1 (latest tag)
- VHDL top, Verilog instance
- Unable to access a verilog signal while being able to access vhdl signals
The difference was that questa actually crashed when trying to read the verilog signal

@cmarqu ``COCOTB_LOG_LEVEL=DEBUG`` helped a lot thanks.

1. So first I did not have ``GPI_EXTRA`` environnement variable set to ``vpi``. Adding it before launching the sim helped me go from 
    ```
    AttributeError: vhdl_top contains no object named verilog_instance
    ```
    to Questasim finding the verilog_instance but crashing when retrieving the signal inside it.

2. By adding some LOG_DEBUG I found that the code that was crashing questa is this one:
https://github.com/potentialventures/cocotb/blob/42e72ba3691685e985ff976c62a413b26a3e6f74/lib/fli/FliImpl.cpp#L305-L317
    specifically ``mti_FirstLowerRegion(parent->get_handle<mtiRegionIdT>());``, for, me, is making questa crash. So I commented those lines and on retry I had no more crash. But still, the verilog signal was not found.

3. I then noticed in the debug log that the Fli and Vpi did not print the same name for the signal they where trying to find in questa. Fli tried finding ``/top_vhdl/verilog_instance/signal`` and Vpi tried ``/top_vhdl.verilog_instance.signal``. Thus I changed the following line:
https://github.com/potentialventures/cocotb/blob/42e72ba3691685e985ff976c62a413b26a3e6f74/lib/vpi/VpiImpl.cpp#L248
    replacing the ``"."`` by a ``"/"``

4. I used to have a vhdl_instance instead of a verilog one and I had some signals capitalized in python but not in the hdl code, which worked for VHDL but obviously not for Verilog (case sensitive) so I had to change that haha.

Hopefully this helps someone with one or more issues in common, cause I think I got them all haha.
I am very much not convinced about part 3's solution though, and I have no idea why it was made with a dot in the first place.
Also where did the /lib directory go in the master branch ??

cmarqu, 2019-07-15:
``lib/`` in master is at https://github.com/potentialventures/cocotb/tree/master/cocotb/share/lib

mmdems, 2019-07-24:
> I have no idea why it was made with a dot in the first place

Maybe it's because of `modelsim.ini` file. If you have a look at lines 998-1003 then you will see:
```
; Specify whether paths in simulator commands should be described
; in VHDL or Verilog format.
; For VHDL, PathSeparator = /
; For Verilog, PathSeparator = .
; Must not be the same character as DatasetSeparator.
PathSeparator = /
```
Maybe these thing are connected somehow, I dont know

But I still cant to get the solution. GPI and VPI are loaded both (at least I can see it in log) so I dont need to specify `GPI_EXTRA`. And also I tried to manipulate with symbols replacing and lines commenting as you recommended, but it seems that I have no any results. But thanks for your reply anyway.

mmdems, 2019-08-09:
I think I'm coming closer to solution. After some experiments I found that one method gave me what I want. I talk about `_discover_all() ` method. And code is:

```
import cocotb
from cocotb.triggers import Timer

@cocotb.test()
def test_mux(dut):
    """Test description"""

    yield Timer(5000)
    dut.d_i.value, dut.sel.value = 1101, 10
    yield Timer(5000)

    dut.mux2to1_1._discover_all()
    dut._log.info(dut.mux2to1_1.i_d.value)
    dut._log.info(dut.mux2to1_1.o_d.value)

    yield Timer(10000)
```

After this I can finally see values of Verilog instances in `_log.info`. I'm not sure, but I think I have to use this method every time when crossing 'language domain'

cmarqu, 2019-08-12:
So you are saying that after you run ``_discover_all()``, these "magically" come into existence? That's interesting, and a good find which should help making it work properly without the need for this call at some point in the future.

TrampHRC, 2020-02-18:
It's been a while but I'm back at it with this issue.
The `_discover_all()` fix works for me too.
It actually searches for children using different functions than the `HierarchyObject.__getattr__` function (found in [cocotb.handle](https://cocotb.readthedocs.io/en/latest/_modules/cocotb/handle.html)). It builds a cache, which will then be used by the  `HierarchyObject.__getattr__` function, instead of its normal way of search. So the "magic" is just the cache building which uses different set of tools, which actually find Verilog signals.

My money is still on the VpiImpl::native_check_create function.
 [https://github.com/cocotb/cocotb/blob/42e72ba3691685e985ff976c62a413b26a3e6f74/lib/vpi/VpiImpl.cpp#L245-L282](https://github.com/cocotb/cocotb/blob/42e72ba3691685e985ff976c62a413b26a3e6f74/lib/vpi/VpiImpl.cpp#L245-L282)

I'm finding it very hard to investigate since I can't find how to make the LOG_DEBUG functions in the C code print. I used to do it but I can't remember how... Can somebody help ?

eric-wieser, 2020-02-18:
> since I can't find how to make the LOG_DEBUG functions in the C code print. I used to do it but I can't remember how... Can somebody help ?

Have a look in cocotb/log.py, I think that calls a function to change the C log level, be possibly conditional on an env car.

TrampHRC, 2020-02-19:
I already tried setting `COCOTB_LOG_LEVEL` to `DEBUG` but it did not work. Only Python debug logs get printed, not C code logs. I even tried tweaking the gpi_logging.c file but I am no C master and it did not help.

eric-wieser, 2020-02-19:
You could try changing these lines:
https://github.com/cocotb/cocotb/blob/1e8880ebc0a15b486d08d7e81091c4fafcfd0293/cocotb/share/lib/gpi_log/gpi_logging.c#L36

https://github.com/cocotb/cocotb/blob/1e8880ebc0a15b486d08d7e81091c4fafcfd0293/cocotb/share/lib/gpi_log/gpi_logging.c#L110

It's possible that the log you're hoping for occurs before python has read the env var.

stezskal, 2020-12-03:
I was also stuck on traversing hierarchy between vhdl top and verilog sub-module with similar issues... Modelsim segmentation fault and error related to Memory Allocation.

Once I added the dut.[verilog module]._discover_all() this also resolved my issue.  Thanks for sharing.