cocotb/cocotb #1006: Simulation fails after initialization `Failed to get the _logFromC method#`

lcapossio, 2019-07-09:
I'm experiencing a problem in Modelsim possibly due to `https://github.com/potentialventures/cocotb/pull/984`. Simulation fails after initialization with the following:
```
Traceback (most recent call last):
  File "C:\ProgramData\Miniconda3\lib\site-packages\cocotb\log.py", line 150, in __getattr__
    return getattr(self.logger, attribute)
AttributeError: 'SimBaseLog' object has no attribute '_logFromC'
Failed to get the _logFromC method#
```

I'm running on Windows 10. Cocotb commit `a5ce571e9abf76b8d4bd89ab9ce579544dc6c762`

eric-wieser, 2019-07-09:
Note that if you don't put `` ` `` around a commit, github links it: a5ce571e9abf76b8d4bd89ab9ce579544dc6c762

eric-wieser, 2019-07-09:
It sounds like you have a bad installation - `SimBaseLog._logFromC` is defined here:

https://github.com/potentialventures/cocotb/blob/a5ce571e9abf76b8d4bd89ab9ce579544dc6c762/cocotb/log.py#L56-L77

Your error message says

> C:\ProgramData\Miniconda3\lib\site-packages\cocotb\log.py", line 150, in __getattr__

But `__getattr__` doesn't even exist in https://github.com/potentialventures/cocotb/blob/master/cocotb/log.py, as it was removed in 32a2fed7faee9e066d6c8284da3675b0cfbfdbe5.

imphil, 2019-07-09:
This looks like a local installation problem. @lcapossio please close this issue if you find the root cause.

lcapossio, 2019-07-09:
Yes it seems like an installation problem. It has to do with how cocotb is installed in windows. I followed the procedure here: https://github.com/potentialventures/cocotb/pull/882. And then @themperek suggest I did a `pip install cocotb` which made it work for a while, but not for the new commits. It seems the package from pip doesn't follow the latests commits in the repo, presumably because they only update it on every official release. Then there is a mismatch with the local checked out repo and the installed package, but I'm unsure why in order for it to work I have to have both in the first place. Probably @themperek will know.

eric-wieser, 2019-07-10:
You should run `pip uninstall cocotb`, `pip install -e path/to/local/cocotb`

lcapossio, 2019-07-10:
That is the way to do it, thanks case closed.

closed 2019-07-10 by lcapossio