cocotb/cocotb #1008: Package for protocols

themperek, 2019-07-09:
At some point, we decided to separately package protocols like Amba/Avalon/opb.

There is still a question should every protocol have a separate repository of it would be part of main cocotb repository (in the dedicated folder)?

cmarqu, 2019-07-11:
VUnit had the same question to answer recently. Their discussions about this (e.g. licenses) could be useful to us too. Here is roughly the start of it: https://gitter.im/VUnit/vunit?at=5cf3ce9b6fc5846bab4ca190

themperek, 2019-07-12:
Some arguments for the separated repository:
- can assign different maintainers
- can be more involved (more dependency on external packages like cocotb-coverage, numpy, etc..)
- can do separate CI

    Should be under placed  https://github.com/cocotb if possible. Should have listed all in documentation/READMY?

Against:
 - more complex maintenance

Anything more?

wallento, 2019-07-17:
I think now that everything is even more Pythonic we should have them as separate modules and repositories. I find the sphinx infrastructure quite efficient, where a few extensions reside in the main tree, while the majority just shares the namespace. I tried to replicate this in my cocomod-uart and cocomod-fifointerface, where cocomod is a kind of unelegant working title ;)

mciepluc, 2019-07-31:
Given the fact, that protocol-specific work may, at the end of the day, expand to standalone complex "Verification IP", and also you may imagine the situation that multiple versions of such VIPs may exist, it is definitely better to keep them in a separate packages. 
I have a dream that one day commercial versions of cocotb-based VIPs will also be available :)

wallento, 2019-07-31:
But they should be uniquely named, right? Going along with a default namespace seems like a good in my opinion and it works quite well in Sphinx. It differentiates the packages in the Python world and streamline the usage and distribution a bit.

Martoni, 2019-10-09:
Is there a «guideline» on how to write a library protocol for cocotb ?
If not, it could be a good start for developing a library «network» for Cocotb no ?

themperek, 2019-10-09:
I do not know of any guidelines. 

Do you consider some particular network protocols?

Martoni, 2019-10-09:
Not really network in fact but i'm using a SPI bus protocol for my actual project.

cmarqu, 2019-10-09:
FWIW, I know @sjalloq has worked on an SPI (master) too.

themperek, 2019-10-09:
If there is interest in SPI (and willingness to work on) maybe we do separate issue? One way this can be approached is to look at what exists in UVM and "make it better"?

Martoni, 2019-10-09:
>  If there is interest in SPI (and willingness to work on) maybe we do separate issue?

Yes definitely. The question was mainly about a "guideline" document on how to write an extension.

wallento, 2019-11-13:
Hi,

I have commented in #1141 about it, but its probably cleaner here.
We should start moving packages out of the core cocotb repository. The proposal is to use namespaces similar to sphinx, which uses `sphinxcontrib.*` as namespace for modules.

@eric-wieser proposed to use `cocotb.ext.*`, but that doesn't work unless we get rid of the cocotb `__init__.py` which will break compatibility.

I used `cocomod.*` in my proposal and I believe we should encourage use of Python3 built-in namespaces, alternatively we can use 'pkg_util-style', see https://packaging.python.org/guides/packaging-namespace-packages/

So, summarized, I understand that we can rule out anything in the cocotb.* namespace outside the core repository (due to the init file). So, what are proposals for another shared namespace?

Overall this is how this should work:

- Packages with cocotb modules declare themselves in a shared namespace, such as `cocomod`
- Packages should be published as `cocomod-<name>`
- Packages with certain quality shall be moved into the github organization, with their specific maintainers, loosening the review pressure on the core maintainers

As said, not emotional about cocomod ;)

Cheers,
Stefan

imphil, 2020-01-31:
We're one step further with #1344 documenting how to write extensions and what naming we expect.

An approach to the repo issue is in https://github.com/cocotb/cocotb/issues/1360, pleaes follow up there as needed.

commit 860f585d (2020-02-01): Document extension best practices

Document agreed-on community best practices for writing cocotb
extensions.

Fixes #1008
closed 2020-02-01 by imphil