cocotb/cocotb #1014: vpiStringVar cannot be assigned anymore

cmarqu, 2019-07-11:
I only now realized that the support for assigning Verilog strings (https://github.com/potentialventures/cocotb/pull/785) has been broken by PR
https://github.com/potentialventures/cocotb/pull/792. 
This has gone unnoticed because the test for var strings (in ``test_discovery``) doesn't run in our CI.
When I change the code back to use ``vpiNoDelay`` then the test passes again (and also my ``mixed_signal`` branch starts working which is where I noticed the problem in the first place).
According to @eric-wieser it cannot just be changed back though "because it means the behavior of ``.setimmediatevalue`` depends on whether the value fits in a string."

eric-wieser, 2019-07-11:
It's possible that all of the `vpiNoDelay` can be replaced with `vpiInertialDelay` - my comment was just that it didn't make much sense for different delays to be used depending on how a binaryvalue was represented, and I picked the one that had a majority vote.

Do you know why inertialDelay would cause a problem?

cmarqu, 2019-07-11:
No, I don't know why inertial would not work. It may even be a simulator bug? I'm testing with Cadence (whether IUS or Xcelium did not make a difference).

cmarqu, 2019-07-13:
I have googled around and grepped the Cadence and Mentor VPI examples and I have only ever seen ``vpiInertialDelay`` used when there was some real delay to be modelled, otherwise it was all ``vpiNoDelay``. Using ``vpiNoDelay`` for everything in cocotb however did not work ("FATAL: We are calling up again"), so I only used it for cases when ``vpiStringVar`` is being assigned in #1016.

PR #1016 (2019-07-16): Use vpiNoDelay flag for vpiStringVar assignments.
Fixes #1014.
closed 2019-07-16 by imphil