cocotb/cocotb #1017: fprintf messages in C code do not give context

cmarqu, 2019-07-13:
I got the error "FATAL: We are calling up again" (while investigating #1014) but nothing hinted at this being a message from cocotb, much less any additional context like file name or line number.
We should at least start such strings with "cocotb: ".

eric-wieser, 2019-07-13:
I think this is because it did not come from cocotb, but from within the simulator. My guess is that writing a signal with no delay causes the edge event to fire on the same internal tick.

cmarqu, 2019-07-13:
This specific string comes fromhttps://github.com/potentialventures/cocotb/blob/a8765a6c6e41acb3e7c09fc61d5514cdd8d86608/cocotb/share/lib/utils/cocotb_utils.c though.

cmarqu, 2019-07-19:
Should we just prefix "cocotb: " here (for the rare cases where such messages are actually printed), or is there an easy way of hooking this up into some logging mechanisms that provides context?

closed 2019-07-21 by themperek