cocotb/cocotb #1024: VPI: Explore always using vpiNoDelay

imphil, 2019-07-16:
Currently we're using vpiInertialDelay to set signals (except for strings). vpiNoDelay could be more appropriate, but doesn't work currently.

Explore if switching makes sense, and what roadblocks we might hit.

(Discussion started in https://github.com/potentialventures/cocotb/issues/1014 and https://github.com/potentialventures/cocotb/pull/1016)

mballance, 2020-04-21:
I echo @cmarqu's comments in #1016: using inertial delays isn't common in my experience. I worked developing CPU simulation models ~20 years ago on multiple simulators (there were quite a few Verilog simulators back then), and we never used inertial delays.
I tried using vpiNoDelay with cocotb master and a new-ish version of Questa, and it seemed to work fine. Are there specific simulators that seem to have issues?

garmin-mjames, 2020-11-06:
Icarus (as of 12.0 (devel) (s20150603-954-g359b2b65)) calls `cbValueChange` callbacks immediately when using `vpiNoDelay`: https://github.com/cocotb/cocotb/issues/1484#issuecomment-722592589

toddstrader, 2024-03-18:
Is it still desired to abandon `vpiInertialDelay` and move entirely to `vpiNoDelay`?  I'm reading through the backstory here (#1484 , #1662 , #1014 , #768 , #2178 ) and it seems like this has been a long and winding road.

Also, is it known if there are currently issues with simulators other than Icarus?

ktbarrett, 2024-03-18:
I don't think it really should matter since we already wait until the read write stage to apply writes. It does, however, make `setimmediatevalue` function weirdly. I don't really understand the original motivation to use `vpiNoDelay`. I think this was more of an investigation into how well it would work or if it were even a good idea.

> Explore if switching makes sense, and what roadblocks we might hit.

I don't anticipate this changing any time soon.

toddstrader, 2024-03-18:
Gotcha, so can you elaborate on what the desired behavior is?  `38.34 vpi_put_value()` says:
```
vpiInertialDelay All scheduled events on the object shall be removed before this event is
scheduled.
```
which I see causes a `vpiSchedEvent` to be created even if there's no other obvious (at least to me) event scheduled on the object in question:
https://edaplayground.com/x/aDri

So my read on `vpiInertialDelay` is that it means "don't apply this value until the simulator's scheduler has moved forward in some way".  If that's the case, how is this different than using a handle's `value` as opposed to `setimmediatevalue`?  I thought `value` was the "non-blocking" assignment and `setimmediatevalue` was the "blocking" one.  But with `vpiInertialDelay` there doesn't appear to truly be a way to be "blocking".  Well, you could use `Force`, but then you'd have to immediately follow it up with a `Release` if what you really wanted was a `Deposit`.  That seems a little wonky and unnecessarily slow.

ktbarrett, 2024-03-18:
> I thought `value` was the "non-blocking" assignment and `setimmediatevalue` was the "blocking" one.

This was the intent AFAICT.

> But with `vpiInertialDelay` there doesn't appear to truly be a way to be "blocking".

Right... `setimmediatevalue` is broken IMO. I think some effort could be done to change this, with the understanding we are doing the right thing, but breaking Icarus.

I'm actually kinda suspicious about the behavior of `.value` sets now. If we really want them to act like non-blocking/signal assignments, they should be applied immediately but inertially, not in `ReadWrite`. This takes me back to #3544.

toddstrader, 2024-03-19:
So are you saying that:
* `.value` should use `vpiInertialValue` and not go through `Scheduler._write_calls`
* `setimmediatevalue` should use `vpiNoDelay`

I'm not sure what the Icarus implications are here, but I do know that the above changes ^ would break Verilator as it currently doesn't observe the `flags` parameter to `vpi_put_value()`.  To address that we could either:
* keep `Scheduler._write_calls` but just for Verilator
* implement deferred VPI puts in `verilator.cpp`

I suggest the latter instead of changes to Verilator itself because it doesn't really understand things like the ReadWrite and ReadOnly phases.  They're more a construct of how cocotb is using Verilator.  We could possibly shoehorn something in Verilator to solve this too, but I'm not sure if that's the best path.

ktbarrett, 2024-03-21:
> So are you saying that:
> 
> * `.value` should use `vpiInertialValue` and not go through `Scheduler._write_calls`
> * `setimmediatevalue` should use `vpiNoDelay`

The 1st I'm still iffy on. The second, yes absolutely.

> To address that we could either:
> 
> * keep `Scheduler._write_calls` but just for Verilator
> * implement deferred VPI puts in `verilator.cpp`

I'm actually thinking we could apply writes just before we return to the simulator in "normal" mode rather than at the beginning of `ReadWrite`. This achieves the intended effect and reduces the number of calls into cocotb which should ~1.5-2x performance of cocotb overall. We would leave the `_write_call` queue and what not, just apply them at a different time.

ktbarrett, 2024-04-03:
@toddstrader that sounds good to me.

marlonjames, 2024-04-03:
We need a way to handle the icarus callback that will occur with `vpiNoDelay`.

toddstrader, 2024-04-03:
> We need a way to handle the icarus callback that will occur with `vpiNoDelay`.

I'm guessing this has something to do with why the `issue_768_*` tests are failing on me.  But I'm not sure why.  @marlonjames can you elaborate on `the icarus callback`?

marlonjames, 2024-04-03:
If there are any `Edge`, `RisingEdge`, or `FallingEdge` triggers waiting on the signal that you write to with `vpiNoDelay`, icarus will call the value change callback immediately, while cocotb is still running.
This currently causes a failure because cocotb expects to return to the simulator before the next callback occurs.

We can catch it and either queue up the callback in the GPI, or queue up the trigger in the cocotb scheduler, but we don't do that right now.

I linked this above, but here is where we initially saw this: https://github.com/cocotb/cocotb/issues/1484#issuecomment-722592589
I have an issue for callbacks that are called while inside a callback here: https://github.com/cocotb/cocotb/issues/2178

toddstrader, 2024-04-03:
OK, cool.  Thanks for the explainer.  I'm currently seeing this:
```
     0.00ns INFO     cocotb.regression                  issue_768.test failed
                                                        Traceback (most recent call last):
                                                          File "/usr/scratch/local/devs/cocotb-stock/tests/test_cases/issue_768_a/issue_768.py", line 21, in test
                                                            assert dut.stream_in_data.value == 0
                                                        AssertionError: assert LogicArray('ZZZZZZZZ', Range(7, 'downto', 0)) == 0
                                                         +  where LogicArray('ZZZZZZZZ', Range(7, 'downto', 0)) = LogicObject(sample_module.stream_in_data).value
                                                         +    where LogicObject(sample_module.stream_in_data) = HierarchyObject(sample_module).stream_in_data
```
which, at least on the face of it, doesn't smell like a reentrancy problem.  It's just like the `setimmediatevalue()` doesn't stick.  I'll re-read those issues and see if anything discussed there is happening in this test now and / or dig in further to see what's going on.

Also, I don't think anything in those 768 tests awaits on edges of the signal that's being set with `vpiNoDelay`.  The only `await`s in there are on `Timer`s and `ReadOnly`.

PR #4068 (2024-08-28): Fix setimmediate
Depends on #3861. Alternative to #3825. Closes #1024.
closed 2024-08-28 by ktbarrett