cocotb/cocotb #1028: Compile sources into different libraries

mmdems, 2019-07-17:
Hi, is it possible to compile 2 sources into 2 different libraries?

For example, if I have 2 files:
```
VHDL_SOURCES=  $(HOME)/cocotb_test/src/my_module.vhd
VHDL_SOURCES+= $(HOME)/cocotb_test/src/my_utils.vhd
```
And I want to compile them to `my_module_lib` and `my_utils_lib` correspondingly. I saw #516 , but it seems, that it has no effect. What do I need to do this?

I think the problem partially is in compile sequence, because if you have a look at the log, then you will see that compiler compiles 2 files in "1 stage":
`vcom -work work ../src/my_module.vhd ../src/my_utils.vhd`

And all I need is obviously is something like this:
`vcom -work my_module_lib ../src/my_module.vhd`
`vcom -work my_utils_lib ../src/my_utils.vhd`

themperek, 2019-07-18:
The only way I can think of currently (for Questa) is to make a custom script and define `SCRIPT_FILE` (https://github.com/cocotb/cocotb/blob/73be2e97947e105c2c091f2f029208d6f5e743cf/cocotb/share/makefiles/simulators/Makefile.questa#L104).

The other way is to try to use this: https://github.com/themperek/cocotb-test and make full own custom script but this is experimental still.

mmdems, 2019-07-18:
> The only way I can think of currently (for Questa) is to make a custom script and define `SCRIPT_FILE`

Thanks, works for me. I close it

closed 2019-07-18 by mmdems