cocotb/cocotb #1033: Forked coroutine _monitor_recv raised exception: 'RisingEdge' object attribute 'log' is read-only

cmarqu, 2019-07-18:
With the ``test_dff`` testcase and ``TOPLEVEL_LANG=vhdl`` and current master I am getting the following exception:
```
     0.00ns INFO     Starting test: "run_test_001"
                     Description: Automatically generated test
                     
                     
     0.00ns ERROR    Forked coroutine _monitor_recv raised exception: 'RisingEdge' object attribute 'log' is read-only
                     Traceback (most recent call last):
                       File "/home/cmarqu/.local/lib/python3.7/site-packages/cocotb/scheduler.py", line 698, in schedule
                         result = coroutine._advance(send_outcome)
                       File "/home/cmarqu/.local/lib/python3.7/site-packages/cocotb/decorators.py", line 164, in _advance
                         raise CoroutineComplete()
                     cocotb.decorators.CoroutineComplete
                     
                     During handling of the above exception, another exception occurred:
                     
                     Traceback (most recent call last):
                       File "/home/cmarqu/.local/lib/python3.7/site-packages/cocotb/scheduler.py", line 489, in unschedule
                         coro.retval
                       File "/home/cmarqu/.local/lib/python3.7/site-packages/cocotb/decorators.py", line 130, in retval
                         return self._outcome.get()
                       File "/home/cmarqu/.local/lib/python3.7/site-packages/cocotb/outcomes.py", line 54, in get
                         raise self.error
                       File "/home/cmarqu/.local/lib/python3.7/site-packages/cocotb/decorators.py", line 154, in _advance
                         return outcome.send(self._coro)
                       File "/home/cmarqu/.local/lib/python3.7/site-packages/cocotb/outcomes.py", line 51, in send
                         return gen.throw(self.error)
                       File "/home/cmarqu/tools/cocotb/examples/dff/tests/dff_cocotb.py", line 56, in _monitor_recv
                         yield clkedge
                       File "/home/cmarqu/.local/lib/python3.7/site-packages/cocotb/scheduler.py", line 531, in _coroutine_yielded
                         trigger.prime(self.react)
                       File "/home/cmarqu/.local/lib/python3.7/site-packages/cocotb/triggers.py", line 280, in prime
                         raise_error(self, "Unable set up %s Trigger" % (str(self)))
                       File "/home/cmarqu/.local/lib/python3.7/site-packages/cocotb/result.py", line 52, in raise_error
                         obj.log.error("%s\n%s" % (msg, buff.getvalue()))
                       File "/home/cmarqu/.local/lib/python3.7/site-packages/cocotb/utils.py", line 589, in __get__
                         setattr(obj, self.fget.__name__, value)
                     AttributeError: 'RisingEdge' object attribute 'log' is read-only
```
(Exception reporting works great now, thanks!)

eric-wieser, 2019-07-18:
Looks like the `__slots__` change missed the log attribute

cmarqu, 2019-07-22:
I tried to fix this myself but wasn't able to. :/

eric-wieser, 2019-07-23:
Adding `"log"` to `__slots__` should solve the problem

cmarqu, 2019-07-23:
This results in ``ValueError: 'log' in __slots__ conflicts with class variable``

imphil, 2019-07-23:
Looks like we need this for 1.2.

eric-wieser, 2019-07-23:
Hmm, that's frustrating. Adding `__dict__` to the slots will fix it too, but might negate the benefit of having `__slots__` in the first place.

The other option is just to revert that commit.

eric-wieser, 2019-07-24:
Note that even after fixing this in #1045 or #1043, this code path should never be reached - it will be interesting to see what the real traceback is (and whether it escapes)

closed 2019-07-24 by imphil