cocotb/cocotb #1046: PyPi deployment not working

imphil, 2019-07-24:
"Skipping a deployment with the pypi provider because this branch is not permitted: v1.2.0"

https://travis-ci.org/cocotb/cocotb/jobs/563023042

@themperek do you know what's going on here? Can you manually upload a package to PyPi to get this release out of the door?

themperek, 2019-07-24:
Will check/do asap (in few hours)

themperek, 2019-07-24:
Seems like "on: tag" is necessary (i have checked few version on my private Travis).

Best would be to change remove "on: tag" realise/tag and do it one more time.
Should I do PR?

imphil, 2019-07-24:
We can try the automation for the next release, but this release is tagged and pushed, we cannot re-tag it.

themperek, 2019-07-24:
Uploaded: https://pypi.org/project/cocotb/

PR #1047 (2019-07-26): Add "tags: true" back to travis DeployPyPI stage
Should fix #1046
closed 2019-07-26 by imphil