cocotb/cocotb #1048: error: Unable to find the root module "top.v" in the verilog source.

vatsalchoksi, 2019-07-24:
Guys, I am trying to run my cocotb testbench with Icarus. I have like 6 verilog modules - all included in one top level module. I have prepared my test and the make file which when I try to execute, I get the following errors.

make results.xml
make[1]: Entering directory '/home/user/cocotb/examples/adder/tests'
/usr/bin/iverilog -o sim_build/sim.vvp -D COCOTB_SIM=1 -s top.v -g2012   /home/user/cocotb/examples/adder/tests/../hdl/top.v
error: Unable to find the root module "top.v" in the Verilog source.
     : Perhaps ``-s top.v'' is incorrect?
1 error(s) during elaboration.
/home/user/cocotb/cocotb/share/makefiles/simulators/Makefile.icarus:61: recipe for target 'sim_build/sim.vvp' failed
make[1]: *** [sim_build/sim.vvp] Error 1
make[1]: Leaving directory '/home/user/cocotb/examples/adder/tests'
/home/user/cocotb/cocotb/share/makefiles/Makefile.sim:84: recipe for target 'sim' failed
make: *** [sim] Error 2

Any idea what is causing the elaboration error?

vatsalchoksi, 2019-07-24:
I am trying to look into the Makefile.icarus where its written.

COMPILE_ARGS += -g2012 # Default to latest SystemVerilog standard

# Compilation phase
$(SIM_BUILD)/sim.vvp: $(SIM_BUILD) $(VERILOG_SOURCES) $(CUSTOM_COMPILE_DEPS)
	$(CMD) -o $(SIM_BUILD)/sim.vvp -D COCOTB_SIM=1 -s $(TOPLEVEL) $(COMPILE_ARGS) $(EXTRA_ARGS) $(VERILOG_SOURCES)

Is there anything that can be done differently in this case?

cmarqu, 2019-07-24:
``TOPLEVEL`` should only be ``top`` I assume, not ``top.v``.

vatsalchoksi, 2019-07-25:
> `TOPLEVEL` should only be `top` I assume, not `top.v`.

@cmarqu  thanks for responding back. I changed that and got the error as below.
make results.xml
make[1]: Entering directory '/home/user/cocotb/examples/adder/tests'
/usr/bin/iverilog -o sim_build/sim.vvp -D COCOTB_SIM=1 -s top -g2012   /home/user/cocotb/examples/adder/tests/../hdl/top.v
/home/user/cocotb/examples/adder/tests/../hdl/top.v:43: error: Unknown module type: top_ctrl
2 error(s) during elaboration.
*** These modules were missing:
        top_ctrl referenced 1 times.
***
/home/user/cocotb/cocotb/share/makefiles/simulators/Makefile.icarus:61: recipe for target 'sim_build/sim.vvp' failed
make[1]: *** [sim_build/sim.vvp] Error 2
make[1]: Leaving directory '/home/user/cocotb/examples/adder/tests'
/home/user/cocotb/cocotb/share/makefiles/Makefile.sim:84: recipe for target 'sim' failed
make: *** [sim] Error 2

To elaborate, I am using 5+ files for simulation. I have created a top module and have all the other files instantiated in it. Cocotb test is directly applied to the top level module. Is there anything I am doing wrong?

Any help is much appreciated.

themperek, 2019-07-25:
It looks like it is not related to CocoTb. You should try first compile design without CocoTb (just call iverilog) and resolve issues (include all verilog sources for compilation?).

Since the issue in the title is resolve I will close this. In In case of new issues please make new one.

closed 2019-07-25 by themperek