cocotb/cocotb #1049: How to assign more than one files to VERILOG_SOURCES in makefile?

vatsalchoksi, 2019-07-25:
I am trying to simulate design that has more than 5 verilog modules based on hierarchy. I have created a top module that has the main modules instantiated in it.

What should be the value assigned to the VERILOG_SOURCES in the make file? Should it only be the top level file? or include all the files with space being the delimiter?

I have tried both. And it gives me respective errors as below:

VERILOG_SOURCES = $(WPWD)/../hdl/top.v

Error: 

make results.xml
make[1]: Entering directory '/home/user/cocotb/examples/adder/tests'
/usr/bin/iverilog -o sim_build/sim.vvp -D COCOTB_SIM=1 -s top -g2012   /home/user/cocotb/examples/adder/tests/../hdl/top.v 
/home/user/cocotb/examples/adder/tests/../hdl/top.v:43: error: Unknown module type: top_ctrl
/home/user/cocotb/examples/adder/tests/../hdl/top.v:65: error: Unknown module type: M25
3 error(s) during elaboration.
*** These modules were missing:
        M25 referenced 1 times.
        top_ctrl referenced 1 times.
***
/home/user/cocotb/cocotb/share/makefiles/simulators/Makefile.icarus:61: recipe for target 'sim_build/sim.vvp' failed
make[1]: *** [sim_build/sim.vvp] Error 3
make[1]: Leaving directory '/home/user/cocotb/examples/adder/tests'
/home/user/cocotb/cocotb/share/makefiles/Makefile.sim:84: recipe for target 'sim' failed
make: *** [sim] Error 2

VERILOG_SOURCES = $(WPWD)/../hdl/top.v top_ctrl.v top_shift.v fifo.v counter.v mem.v LC25.v 

Error:

make results.xml
make[1]: Entering directory '/home/user/cocotb/examples/adder/tests'
make[1]: *** No rule to make target 'top_ctrl.v', needed by 'sim_build/sim.vvp'.  Stop.
make[1]: Leaving directory '/home/user/cocotb/examples/adder/tests'
/home/user/cocotb/cocotb/share/makefiles/Makefile.sim:84: recipe for target 'sim' failed
make: *** [sim] Error 2

Can someone help with this? Apologies for the trouble, shifting from SV to this and hence initial struggles.

eric-wieser, 2019-07-25:
You're not specifying the paths correctly:
```
VERILOG_SOURCES = $(WPWD)/../hdl/top.v $(WPWD)/../hdl/top_ctrl.v $(WPWD)/../hdl/top_shift.v $(WPWD)/../hdl/fifo.v $(WPWD)/../hdl/counter.v $(WPWD)/../hdl/mem.v $(WPWD)/../hdl/LC25.v
```
There are definitely makefile tricks that make writing that shorter or more readable

vatsalchoksi, 2019-07-25:
Thanks Eric! If you have any document or link in suggestion regarding makefiles , I would be glad to go through them. I will close this ticket here.

closed 2019-07-25 by vatsalchoksi

cmarqu, 2019-07-25:
You can append to make variables with ``VERILOG_SOURCES += ...``