cocotb/cocotb #1060: Assignments with "<=" and ReadOnly

corna, 2019-08-06:
Hi everyone,

I'm trying to understand the behaviour of cocotb's "<=" assignment.
According to the documentation:

> The syntax sig <= new_value is a short form of sig.value = new_value. It not only resembles HDL-syntax, but also has the same semantics: writes are not applied immediately, **but delayed until the next write cycle**. Use `sig.setimmediatevalue(new_val)` to set a new value immediately [...]

By looking into `scheduler.py`:

> Due to the simulator nuances and fun with delta delays we have the
>    following modes:
>
>    Normal mode
>        - Callbacks cause coroutines to be scheduled
>        - **Any pending writes are cached and do not happen immediately**
>
>    ReadOnly mode
>        - Corresponds to cbReadOnlySynch (VPI) or vhpiCbLastKnownDeltaCycle
>          (VHPI).  In this state we are not allowed to perform writes.
>
>    Write mode
>        - Corresponds to cbReadWriteSynch (VPI) or vhpiCbEndOfProcesses (VHPI)
>          **In this mode we play back all the cached write updates.**
>

From this, my guess is that, every time I perform a write with the _<=_ operator, the write operation gets delayed until the Write mode is reached. However, with [this test case](https://github.com/cocotb/cocotb/files/3472157/cocotb_readonly.zip), the value changes immediately after the `await ReadOnly()`.

```
[...]

dut.inputwire <= 0
await Timer(1, "ns")
dut.inputwire <= 1

print(dut.inputwire.value.integer) # prints 0, as expected

await ReadOnly()

print(dut.inputwire.value.integer) # prints 1, but I was expecting 0

[...]
```

Shouldn't the value be `0` until the Write mode cycle? Did I misunderstood something?
I've checked both with `ghdl` and `questa`, the result is the same.

Thanks.

eric-wieser, 2019-08-06:
The read-only cycle comes after the write-mode cycle, not before

corna, 2019-08-08:
Thanks, that explains the issue.

However I am now facing another issue related to the first one. I have seen that the cocotb's monitors awaits for the RisingEdge and then for the ReadOnly phase, to make sure that every signal has been written.
This causes the problematic situation where, at the rising edge of the clock, both the cocotb's writes and the RTL ones are performed.
In some cases, I found this to be problematic, as in [this test case](https://github.com/cocotb/cocotb/files/3481931/cocotb_dflipflop_delay.zip): here I have a D-FF which has:
 * 100 ps of propagation delay in the 0->1 transition
 * 0 ps of propagation delay in the 1->0 transition

Then, I setup a clock in cocotb and I drive the D input with a square wave, while reading the Q output, generating so:

[image]

In cocotb I expect an oscillating output, however, with this particular situation, I always read 0, as, on the clock RisingEdge, the `await ReadOnly()` instructions "anticipates" the write.

This simple test case reflects however some possible issues with the monitors, when there are delays written in the HDL files.

What do you think? Thanks.

eric-wieser, 2019-11-19:
I no longer remember the details, but the way to fix this would be to use the newer VPI callbacks that fire after the last write has finishes, but while still in the readwrite stage

corna, 2019-11-29:
Do you have any reference for those VPI callbacks?

themperek, 2020-04-16:
Actually your [case](https://github.com/cocotb/cocotb/files/3481931/cocotb_dflipflop_delay.zip)   results depend on the simulator. 
If in your replace  `await ReadOnly()` with await `Timer(1, units="ns")` it is OK but I do not know all the quirks of FLI/VHPI.
If you skip  `await ReadOnly()` it also works effectively sample data **before** `RisingEdge()`.

ktbarrett, 2020-07-10:
xref #1655

ktbarrett, 2024-07-25:
We no longer have monitors. 

The difference is the difference in semantics of monitoring. Does the transaction occur when the transition leaves the source (`ReadOnly()`) or when it arrives at the sink (next `RisingEdge(clk)`)?

Also the idiom of
```python
await RisingEdge(clk)
await ReadOnly()
```
Is necessary to get the monitors working with Verilator. Verilator's `RisingEdge(clk)` is essentially a `RisingEdge(clk)` followed by a `ReadWrite`, since it always evaluates the entire design.

closed 2024-07-25 by ktbarrett