cocotb/cocotb #1062: Integrate cocotb-test into cocotb

themperek, 2019-08-08:
I would like to know if there is consensus to merge [cocotb-tests](https://github.com/themperek/cocotb) in current form into cocotb?

In case you would like to see this in should I prepare one big PR or split somehow into smaller pieces?

``cocotb-test`` provides standard python unit testing capabilities 
- compiles the libraries during setup/installation phase using setuptools (separate directory for every simulator) 
- uses python to start simulators (remove the need for Makefiles), includes Makefile compatibility mode
- provides simple pytest plugin to merge results
- allow the look and feel of Python unit testing
- allow easy customization of simulation flow
- easy installation (especially on Windows)
- can create wheel package (no need to compile libraries by user)
- allow to use [pytest-xdist](https://pypi.org/project/pytest-xdist/) / [pytest-parallel](https://github.com/browsertron/pytest-parallel)

Somehow releted: #549 #437 #428 #354 #1057 #991 #960 #807 #482 #595 #711 #696 #667 #508 #528 #498 #488 ...

TC01, 2019-08-08:
I haven't gotten the chance to try cocotb-test yet, but one question:

> compiles the libraries during setup/installation phase using setuptools (separate directory for every simulator)

Does this mean that I can't install the libraries for a simulator I don't have installed on my system?

I think it might be nice to still support the current way of doing things, where the libraries are built at run-time rather than at installation-time. Maybe this falls under Makefile compatibility mode?

I was mainly thinking here about possibly making packages for Linux distributions (for instance, I've been planning to make a package for Fedora/RHEL now that cocotb 1.2 is out). A distribution won't be able to ship libraries compiled against nonfree simulators, but it could ship the source code for all the simulator libraries and let the user build each one on demand (as things work at the moment).

Or I guess as another example... if the libraries are built at installation time, what happens if I need to upgrade the simulator version? (Or if there are two versions of a simulator installed side-by-side on a shared machine and users are able to switch between them). Do I need to reinstall cocotb to recompile the libraries?

I agree that having the option to install prebuilt libraries is nice though. So it might be nice to support both, if possible...

themperek, 2019-08-08:
@TC01 One has to call `pip install` to recompile (if required). Actually for simulators I have (no Aldec) on Linux only FLI require to link something (https://github.com/themperek/cocotb-test/blob/master/cocotb_test/build_libs.py). There is way to recompile without install (can make it bit easier if required and add `cocotb-build-libs` command).

TC01, 2019-08-08:
> One has to call pip install to recompile (if required).

Hm. To me, this seems like something that is potentially a problem for non-pip installs (like a potential distro package), and/or for system-wide installations of cocotb on a server where a user might not have the permissions to recompile the libraries.

Is it possible to still support having the "build" directory locally, as is currently the case?

themperek, 2019-08-08:
>or for system-wide installations of cocotb on a server where a user might not have the permissions to recompile the libraries.

Can do `pip install --user`

>Is it possible to still support having the "build" directory locally, as is currently the case?
Yes. Can do a command `cocotb-build-libs` or make some parameter (variable) to compile locally as is now.

markmelvin, 2019-11-19:
Is there any near-term plan to make cocotb more pytest friendly? I'm about to give cocotb-tests a go, but if it will eventually be integrated in the code base I would be interested in trying that.

eric-wieser, 2019-11-19:
I think there are two different approaches here. `cocotb-test` is a wrapper around the entirety of cocotb, which runs cocotb inside pytest. For this reason, I don't think we should merge the entirety of `cocotb-test` into `cocotb`, in the same was as `pytest-cov` is not merged into `coverage`.

Independently, I'd like to make it possible to host pytest inside cocotb, which ultimately requires making the `RegressionManager` swappable for another test runner.

markmelvin, 2020-01-02:
Having been buried in `pytest` and `cocotb` (and various combinations thereof) for about a week now, I agree that the better approach would be to have a `RegressionManager` that uses `pytest` within `cocotb` itself.

eric-wieser, 2020-01-29:
> like a potential distro package

Wouldn't a distro package be pre-built?

TC01, 2020-01-29:
If we can build all the libraries with only the open-source simulators available, then sure, a distro package could ship pre-built libraries. For some reason I assumed that wasn't the case when I commented earlier.

themperek, 2020-01-29:
So you can look here: https://github.com/cocotb/cocotb/blob/d401fd0f05bdb1887ab7a81e03cec4741e267dfb/cocotb/build_libs.py#L248-L418

We can compile everything but FLI for Modelsim/Questa and Riviera on Linux.

eric-wieser, 2020-01-29:
I think the argument would then go that if the simulator is not built into the distro, then the distro cocotb can't be used with it - the user would need to create a virtualenv with a local cocotb install to pair with their local simulator install.

Does that seem sensible?

TC01, 2020-01-29:
Well, the three use cases I think that should probably be considered are:

1. Personal use of cocotb with open source simulator.
2. Personal use of cocotb with proprietary simulator.
3. Central installation of cocotb on a server for use with proprietary simulator(s).

My feeling is that a distribution package is most useful (to the extent that it's useful at all) for (1) and (3). But if all the sysadmin has to do for (3) is run something like ```sudo cocotb-build-libs questa``` to manually compile Questa support because we can't build that library without Questa present, I think that's fine.

So yes, this seems sensible. :)

eric-wieser, 2020-01-29:
In case 3, is the proprietary simulator also centrally installed? If so, then the sysadmin can just run setup.py too.

themperek, 2022-03-09:
Done in https://github.com/cocotb/cocotb/pull/2634

closed 2022-03-09 by themperek