cocotb/cocotb #1064: Riviera-PRO Windows: Cannot load the "libvpi.dll" library  /dependency issues

mikep996, 2019-08-13:
There is a problem (title) to get working cocotb if you follow the official instructions:
  https://cocotb.readthedocs.io/en/latest/quickstart.html#installing-cocotb
32-bit tools on windows10 64 bit environment.
The  libvpi.dll exist but is corrupted.

I was able to run cocotb it if the libvpi/vhpi.dll has been compiled by gcc version provided with Riviera-PRO.

Is any matter from which the python27.dll is used?
Should I also define (change order) on PATH variable?
thanks

imphil, 2019-08-13:
These instructions are meant only for Modelsim 32 bit. Is Riviera and your compiler really a 32 bit application?

mikep996, 2019-08-14:
Yes, I am using Riviera-PRO 2019.4 32-bit version:
  vsim -ver
  Aldec, Inc. Riviera-PRO version 2019.04.134.7332 built for Windows on May 10, 2019.
Looks the official cocotb instructions are misleading. 

(BTW. If you try to follow instructions and use 'make' command without any arguments you get the info about the lack of Icarus simulator.)

The cocotb quick start provide link to mingw 6.3.0.
Riviera-PRO use own gcc 4.8.4 (Modelsim;  4.2.1)
After manually compiling all dlls. with gcc4.8.4 from Riviera-PRO I was able to execute the endian_swapper example.
This is creepy workaround only. I am looking something more automated.

thanks a lot

cmarqu, 2019-08-14:
@mikep996 There is a actually an open pull request to move this part of the installation instructions into https://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions
Feel free to mention your workaround in a subsection there, maybe someone with Windows experience can then expand/automate it more.

mikep996, 2019-08-16:
done.

PR #1588 (2020-04-07): Improve windows support in Makefiles
Minimal PR to make Widows work with anaconda and msys2 (based on #1462).  
Something like #1565 would also help and keep user code cleaner.

- Set `PYTHONHOME` in order to properly populate `sys.path` in an embedded python interpreter.
- Set `PATH` to include a path to cocotb compiled libraries.
- Set an appropriate path format.
- Cleanup simulator Makefiles (more can be done still).

Tested with Icarus, Aldec Riviera, Questa, Modelsim (32)

Resolves: #1064 #428 #498 #436 #354
closed 2020-04-07 by eric-wieser