cocotb/cocotb #1076: "Unable set up RisingEdge(c) Trigger" with IUS and VHDL

cmarqu, 2019-08-19:
When I run ``examples/dff/tests`` with 
```
make SIM=ius TOPLEVEL_LANG=vhdl COCOTB_LOG_LEVEL=DEBUG COCOTB_SCHEDULER_DEBUG=1 all
```
I am getting the following output:
```
     0.00ns DEBUG    cocotb.regression                         regression.py:555  in generate_tests                  Adding generated test "run_test_001" to module "dff_cocotb"
     0.00ns INFO     cocotb.regression                         regression.py:187  in initialise                      Found test dff_cocotb.run_test_001
     0.00ns INFO     cocotb.regression                         regression.py:321  in execute                         Running test 1/1: run_test_001
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:618  in add                             Adding new coroutine run_test_001
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:695  in schedule                        Scheduling coroutine run_test_001 with outcome Value(None)
     0.00ns INFO     ...test.run_test_001.0x7f4ae5046780       decorators.py:253  in _advance                        Starting test: "run_test_001"
                                                                                                                     Description: Automatically generated test
                                                                                                                     
                                                                                                                     
     0.00ns DEBUG    ...test.run_test_001.0x7f4ae5046780       decorators.py:258  in _advance                        Sending Value(None)
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:701  in schedule                        Coroutine run_test_001 yielded run_test (mode 1)
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:645  in _trigger_from_unstarted_coro    Scheduling nested coroutine: run_test
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:618  in add                             Adding new coroutine run_test
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:695  in schedule                        Scheduling coroutine run_test with outcome Value(None)
     0.00ns DEBUG    cocotb.c                                      handle.py:84   in __init__                        Created
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:618  in add                             Adding new coroutine start
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:695  in schedule                        Scheduling coroutine start with outcome Value(None)
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:695  in schedule                        Scheduling coroutine _do_writes with outcome Value(None)
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:701  in schedule                        Coroutine _do_writes yielded _Event (mode 1)
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:372  in _event_loop                     Trigger fired: _Event
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:434  in _event_loop                     1 pending coroutines for event _Event
                                                                                                                     	_do_writes
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:441  in _event_loop                     Scheduling coroutine _do_writes
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:695  in schedule                        Scheduling coroutine _do_writes with outcome Value(<cocotb.triggers._Event object at 0x7f4ae491f548>)
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:701  in schedule                        Coroutine _do_writes yielded ReadWrite(readwritesync) (mode 1)
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:444  in _event_loop                     Scheduled coroutine _do_writes
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:463  in _event_loop                     All coroutines scheduled, handing control back to simulator
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:701  in schedule                        Coroutine start yielded Timer(2500000.00ps) (mode 1)
     0.00ns DEBUG    cocotb.d                                      handle.py:84   in __init__                        Created
     0.00ns DEBUG    cocotb.q                                      handle.py:84   in __init__                        Created
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:618  in add                             Adding new coroutine _monitor_recv
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:695  in schedule                        Scheduling coroutine _monitor_recv with outcome Value(None)
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:701  in schedule                        Coroutine _monitor_recv yielded RisingEdge(c) (mode 1)
     0.00ns ERROR    cocotb.RisingEdge.0x7f4ae4925c28              result.py:52   in raise_error                     Unable set up RisingEdge(c) Trigger
                                                                                                                     NoneType: None
                                                                                                                     
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:372  in _event_loop                     Trigger fired: NullTrigger()
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:434  in _event_loop                     1 pending coroutines for event NullTrigger()
                                                                                                                     	_monitor_recv
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:441  in _event_loop                     Scheduling coroutine _monitor_recv
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:695  in schedule                        Scheduling coroutine _monitor_recv with outcome Error(TestError('Unable set up RisingEdge(c) Trigger'))
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:714  in schedule                        Coroutine completed: _monitor_recv
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:491  in unschedule                      TestComplete received: TestError
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:754  in finish_test                     finish_test called with TestError('Unable set up RisingEdge(c) Trigger')
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:780  in cleanup                         Killing start
     0.00ns DEBUG    ..tb.coroutine.start.0x7f4ae4924198       decorators.py:182  in kill                            kill() called on coroutine
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:780  in cleanup                         Killing _do_writes
     0.00ns DEBUG    ..routine._do_writes.0x7f4ae4924278       decorators.py:182  in kill                            kill() called on coroutine
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:780  in cleanup                         Killing run_test_001
     0.00ns DEBUG    ...test.run_test_001.0x7f4ae5046780       decorators.py:182  in kill                            kill() called on coroutine
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:444  in _event_loop                     Scheduled coroutine _monitor_recv
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:276  in _check_termination              Test terminating, scheduling Timer
     0.00ns DEBUG    cocotb.scheduler                           scheduler.py:463  in _event_loop                     All coroutines scheduled, handing control back to simulator
     0.00ns INFO     cocotb.scoreboard.dff                     scoreboard.py:216  in add_interface                   Created with reorder_depth 0
[... cut ...]
     1.00ns ERROR    cocotb.regression                         regression.py:300  in handle_result                   Test Failed: run_test_001 (result was TestError)
     1.00ns ERROR    cocotb.regression                         regression.py:207  in tear_down                       Failed 1 out of 1 tests (0 skipped)
```
(I have modified line 695 of ``scheduler.py`` to read ``self.log.debug("Scheduling coroutine {} with outcome {}".format(coroutine.__name__, send_outcome))`` here.)

Any hints as to how to approach debugging this?

eric-wieser, 2019-08-19:
For some reason `simulator.register_value_change_callback` is failing, and unfortunately it doesn't propagate a reason

eric-wieser, 2019-08-19:
I think if you checkout master, you'll get tracebacks pointing to that line?

cmarqu, 2019-08-20:
I am on master, and there is no traceback. In fact, I found that I do not get any of the ``LOG_ERROR`` messages - if I replace them with ``fprintf(stderr, ...)``, I get something.
The code is taking this path: https://github.com/cocotb/cocotb/blob/master/cocotb/share/lib/vhpi/VhpiCbHdl.cpp#L745,
and from ``arm_callback`` I am getting the message "VHPI ERROR: Registered callback isn't enabled! Got 1000".
This in turn is explained by the message from the simulator when run with ``-plidebug``:
```
ERROR:          VHPI       VHUNIMP
         The property or method vhpiStateP has not been implemented yet.
```
So far so bad.
The ``arm_callback`` method for VPI looks a bit different, maybe that logic could also be applied to VHPI...

eric-wieser, 2019-08-22:
>  In fact, I found that I do not get any of the LOG_ERROR messages - if I replace them with fprintf(stderr, ...), I get something.

That's a little alarming - perhaps `LOG_ERROR` is hanging trying to acquire the GIL. Alternatively, it might be outputting to the simulator console, if you're running in gui mode.

>The property or method vhpiStateP has not been implemented yet.

Is this #153 again?

cmarqu, 2019-08-22:
It's not outputting to the simulator console.
And good find, I had forgotten about this ancient issue. The conversation doesn't seem to be very helpful now unfortunately, and my bug report to Cadence that resulted from it went unanswered.

eric-wieser, 2019-08-24:
Logging output is fixed in gh-1086

antho24, 2019-09-10:
I believe this is the same problem as issue #381. I am running into this same issue with IUS.

cmarqu, 2020-12-01:
FWIW, in current master (and this the upcoming version 1.5) this is printed slightly different, i.e.:
```
0.00ns ERROR    cocotb.gpi                                VhpiCbHdl.cpp:448  in arm_callback                    VHPI: Registered callback isn't enabled! Got vhpiStateP=vhpiUndefined(1000)
```

martijnbastiaan, 2021-07-15:
We're encountering this issue when using Xcelium too. Because it supports cross-language compilation, you can write a Verilog shim. This makes cocotb talk to the simulator over VPI which doesn't encounter similar issues as the VHPI path.

## Workaround example
<details>
<summary>Click to expand</summary>

The following cocotb test case + hdl files work fine in GHDL but fail with Xcelium:

**adder.vhdl**:
```vhdl
port
  ( clk : in std_logic
  ; a : in signed(31 downto 0)
  ; b : in signed(31 downto 0)
  ; c : out signed(31 downto 0)
  );
end adder;

architecture Behavioral of adder is
begin
  p_add : process(clk)
  begin
    if rising_edge(clk) then
      c <= a + b;
    end if;
  end process p_add;
end Behavioral;
```

**test.py**:

```python
import cocotb

from cocotb.clock import Clock
from cocotb.triggers import RisingEdge

@cocotb.test()
async def simple_test(dut):
    clock = Clock(dut.clk, 50, units='ns')
    cocotb.fork(clock.start())

    dut.a.value = 5
    dut.b.value = 7

    # Wait for simulator to start
    await RisingEdge(dut.clk)

    # Latch a+b
    await RisingEdge(dut.clk)

    # a+b on output
    await RisingEdge(dut.clk)

    assert dut.c.value.integer == 12
```

**Makefile**:
```makefile
SIM ?= ghdl
TOPLEVEL_LANG ?= vhdl
MODULE := test

VHDL_SOURCES = adder.vhdl

include $(shell cocotb-config --makefiles)/Makefile.sim
```

To work around this, add a Verilog shim:

**adder_shim.v**:
```verilog
module adder_shim
  ( input         clk
  , input  [31:0] a
  , input  [31:0] b
  , output [31:0] c
  );

  adder d0 (clk, a, b, c);
endmodule
```

and change your `Makefile` such that it uses the shim when compiling with  Xcelium:

```makefile
SIM ?= ghdl
TOPLEVEL_LANG ?= vhdl
MODULE := test

VHDL_SOURCES = adder.vhdl

ifneq ($(filter $(SIM),xcelium),)
    VERILOG_SOURCES = adder_shim.v
    TOPLEVEL_LANG := verilog
    TOPLEVEL := adder_shim
else
    TOPLEVEL_LANG := vhdl
    TOPLEVEL := adder
endif

include $(shell cocotb-config --makefiles)/Makefile.sim
```

</details>

Toucanfan, 2023-01-17:
Probably a stupid question, since I have no insight into VHPI, but is there no other way the RisingEdge trigger could be implemented than via the current approach?

marlonjames, 2023-01-17:
We always register callbacks as enabled, so we could assume the state is `vhpiEnable` without checking it.

Toucanfan, 2023-01-17:
I tried removing the check, and the simulation gets started. However, it later fails with the following message:

50.00ns  ERROR  gpi  FATAL: We are calling up again

This is with Xcelium 18.12

marlonjames, 2023-01-18:
You can try setting `COCOTB_LOG_LEVEL` environment variable to `DEBUG` or `TRACE` to get more info, but they will be increasingly verbose, so if a lot happens before that 50ns message you could get a flood of log entries.

You could also attempt to debug with gdb using https://docs.cocotb.org/en/stable/building.html#envvar-COCOTB_ATTACH

Toucanfan, 2023-01-18:
The DUT seems to be correctly discovered, but then this happens:

[image]

I wonder whether this memory corruption error has anything to do with skipping the checks earlier. Seems to coincide (in simulation time) with the first await on RisingEdge.

marlonjames, 2023-01-18:
This looks like a case of callback re-entrancy as detailed in #2178.

The TRACE entry "Passing control to GPI user" happens when the simulator calls a registered GPI callback.
That is happening during another callback.

If you could debug it and put a break point in `handle_vhpi_callback()`, you might get some better insight by inspecting the `vhpiCbDataT` struct.

Toucanfan, 2023-01-18:
I managed to attach GDB and set the breakpoint - however, it seems there is no debug information available in the loaded cocotb libraries, so I can't do "info args", for example. I tried adding "-g" to _extra_cxx_compile_args in cocotb_build_libs.py and do a reinstall with "pip3 --user --no-build-isolation .", but it didn't help.

marlonjames, 2023-01-18:
If you have `nox` installed, the `dev` session installs and editable build with debug info and coverage.

Run `nox -s dev -- bash` from the cocotb repo. That will setup the environment and run bash with everything installed. You can also use a bash alternative.

I've confirmed with Icarus Verilog that the cb_data info is available:

```gdb
(gdb) b handle_vpi_callback
Breakpoint 1 at 0x7f5c6b948100: file cocotb/share/lib/vpi/VpiImpl.cpp, line 629.
(gdb) continue
Continuing.

Breakpoint 1, handle_vpi_callback (cb_data=0x55e859fadcd0) at cocotb/share/lib/vpi/VpiImpl.cpp:629
629     int32_t handle_vpi_callback(p_cb_data cb_data) {
(gdb) where
#0  handle_vpi_callback (cb_data=0x55e859fadcd0) at cocotb/share/lib/vpi/VpiImpl.cpp:629
#1  0x000055e8596e848b in vpiStartOfSim () at vpi_callback.cc:541
#2  0x000055e8596b3e11 in schedule_simulate () at schedule.cc:1154
#3  0x000055e8596649a5 in main (argc=<optimized out>, argv=<optimized out>) at main.cc:469
(gdb) print *cb_data
$2 = {reason = 11, cb_rtn = 0x7f5c6b948100 <handle_vpi_callback(p_cb_data)>, obj = 0x0, time = 0x55e859f2f2e0,
  value = 0x0, index = 0, user_data = 0x55e859f2f2a0 "@_\227k\\\177"}
```

Toucanfan, 2023-01-18:
Ok, I got cocotb compiled with debug info on. Not using nox (since I'm behind a corporate firewall, and downloading packages is a pain), but instead invoking pip as follows:

`pip3 install --user --no-build-isolation --global-option build --global-option --debug .`

Then I attached  GDB and ran the following session:

```
(gdb) b handle_vhpi_callback
Breakpoint 1 at 0x7f9136e84a6d: file cocotb/share/lib/vhpi/VhpiImpl.cpp, line 988.

(gdb) handle SIGSEGV nostop noprint pass
Signal        Stop      Print   Pass to program Description
SIGSEGV       No        No      Yes             Segmentation fault

(gdb) cont
Continuing.
warning: Corrupted shared library list: 0x578d9f0 != 0x578b100

Breakpoint 1, handle_vhpi_callback (cb_data=0x28d6ae0) at cocotb/share/lib/vhpi/VhpiImpl.cpp:988
988         gpi_to_user();

(gdb) print *cb_data
$1 = {reason = 1001, cb_rtn = 0x7f9136e84a61 <handle_vhpi_callback(vhpiCbDataT const*)>, obj = 0x500080 <deposit_doit+6954>, time = 0x5188808, value = 0x0,
  user_data = 0x57c5028}

(gdb) cont
Continuing.

Breakpoint 1, handle_vhpi_callback (cb_data=0x28d6ae0) at cocotb/share/lib/vhpi/VhpiImpl.cpp:988
988         gpi_to_user();

(gdb) print *cb_data
$2 = {reason = 1001, cb_rtn = 0x7f9136e84a61 <handle_vhpi_callback(vhpiCbDataT const*)>, obj = 0x500080 <deposit_doit+6954>, time = 0x5188808, value = 0x0,
  user_data = 0x57c5028}

(gdb) next
990         VhpiCbHdl *cb_hdl = (VhpiCbHdl *)cb_data->user_data;

(gdb) next
992         if (!cb_hdl) {

(gdb) next
998         gpi_cb_state_e old_state = cb_hdl->get_call_state();

(gdb) next
1000        if (old_state == GPI_PRIMED) {

(gdb) next
1001            cb_hdl->set_call_state(GPI_CALL);

(gdb) next
1002            cb_hdl->run_callback();

(gdb) step
GpiValueCbHdl::run_callback (this=0x57c5068) at cocotb/share/lib/gpi/GpiCbHdl.cpp:111
111         std::string current_value;

(gdb) next
114         if (required_value == "X")

(gdb) next
117             current_value = m_signal->get_signal_value_binstr();

(gdb) next
118             if (current_value == required_value) pass = true;

(gdb) next
118             if (current_value == required_value) pass = true;

(gdb) next
121         if (pass) {

(gdb) next
122             this->gpi_function(m_cb_data);

(gdb) step
handle_gpi_callback (user_data=0x57c1060) at cocotb/share/lib/simulator/simulatormodule.cpp:204
204         to_python();

(gdb) step
gpi_log (name=0x7f9135b12dcf "gpi", level=40, pathname=0x7f9135b12da0 "cocotb/share/lib/simulator/simulatormodule.cpp", funcname=0x7f9135b14910 <_ZZ19handle_gpi_callbackPvE8__func__.3170.3180> "handle_gpi_callback", lineno=204, msg=0x7f9135b12d80 "FATAL: We are calling up again") at cocotb/share/lib/gpi_log/gpi_logging.cpp:45
45          va_start(argp, msg);

(gdb) next
46          gpi_vlog(name, level, pathname, funcname, lineno, msg, argp);
48      }
[Inferior 1 (process 22060) exited with code 01]
```

marlonjames, 2023-01-18:
Ok, so both times it is a `vhpiCbValueChange` callback (`cb_data->reason = 1001`).

Both times are also for the same `vhpiHandleT` (`cb_data->obj = 0x500080`).

The part that is important is what happens between the two breakpoints. Are you able to get a stack trace when the breakpoint is hit the second time? For this error I would expect the first callback to be on the stack still.

Toucanfan, 2023-01-18:
Sure, here is the backtrace at each time the breakpoint is hit:

```
+b handle_vhpi_callback
Breakpoint 1 at 0x7f1e2d38fa6d: file cocotb/share/lib/vhpi/VhpiImpl.cpp, line 988.
+handle SIGSEGV nostop noprint pass
Signal        Stop      Print   Pass to program Description
SIGSEGV       No        No      Yes             Segmentation fault
+cont
Continuing.

Breakpoint 1, handle_vhpi_callback (cb_data=0x28d6ae0) at cocotb/share/lib/vhpi/VhpiImpl.cpp:988
988         gpi_to_user();
+bt
#0  handle_vhpi_callback (cb_data=0x28d6ae0) at cocotb/share/lib/vhpi/VhpiImpl.cpp:988
#1  0x000000000082f364 in ?? ()
#2  0x0000000000c2a5a9 in ?? ()
#3  0x0000000000b6c4d8 in ?? ()
#4  0x0000000000b58d88 in ?? ()
#5  0x000000000089a84f in ?? ()
#6  0x00000000008992ca in vhpi_put_value ()
#7  0x00007f1e2d388710 in VhpiLogicSignalObjHdl::set_signal_value (this=0x5d68cc0, value=0, action=GPI_DEPOSIT) at cocotb/share/lib/vhpi/VhpiCbHdl.cpp:527
#8  0x00007f1e33164ad9 in gpi_set_signal_value_int (sig_hdl=0x5d68cc0, value=0, action=GPI_DEPOSIT) at cocotb/share/lib/gpi/GpiCommon.cpp:483
#9  0x00007f1e2c01cad6 in set_signal_val_int (self=0x7f1e2b0a2d70, args=0x7f1e2bff7180) at cocotb/share/lib/simulator/simulatormodule.cpp:657
#10 0x00007f1e2d081395 in cfunction_call_varargs (kwargs=<optimized out>, args=<optimized out>, func=0x7f1e2c0037c0) at Objects/call.c:757
#11 PyCFunction_Call (func=0x7f1e2c0037c0, args=<optimized out>, kwargs=<optimized out>) at Objects/call.c:772
#12 0x00007f1e2d057bb0 in do_call_core (kwdict=0x0, callargs=0x7f1e2bff7180, func=0x7f1e2c0037c0, tstate=0x5770c80) at Python/ceval.c:5007
#13 _PyEval_EvalFrameDefault (f=<optimized out>, throwflag=<optimized out>) at Python/ceval.c:3559
#14 0x00007f1e2d093559 in gen_send_ex (closing=0, exc=0, arg=<optimized out>, gen=0x7f1e2bfe6ac0) at Objects/genobject.c:222
#15 _PyGen_Send (gen=0x7f1e2bfe6ac0, arg=<optimized out>) at Objects/genobject.c:292
#16 0x00007f1e2d0896f5 in method_vectorcall_O (func=0x7f1e2c784ae0, args=0x7f1e2b588880, nargsf=<optimized out>, kwnames=<optimized out>) at Objects/descrobject.c:416
#17 0x00007f1e2d050c4b in _PyObject_Vectorcall (kwnames=<optimized out>, nargsf=<optimized out>, args=<optimized out>, callable=<optimized out>)
    at ./Include/cpython/abstract.h:127
#18 trace_call_function (kwnames=<optimized out>, nargs=<optimized out>, args=<optimized out>, func=<optimized out>, tstate=<optimized out>) at Python/ceval.c:4968
#19 call_function (kwnames=0x0, oparg=<optimized out>, pp_stack=<synthetic pointer>, tstate=0x5770c80) at Python/ceval.c:4984
#20 _PyEval_EvalFrameDefault (f=<optimized out>, throwflag=<optimized out>) at Python/ceval.c:3486
#21 0x00007f1e2d04eca0 in function_code_fastcall (co=<optimized out>, args=<optimized out>, nargs=2, globals=<optimized out>) at Objects/call.c:283
#22 0x00007f1e2d050c4b in _PyObject_Vectorcall (kwnames=<optimized out>, nargsf=<optimized out>, args=<optimized out>, callable=<optimized out>)
    at ./Include/cpython/abstract.h:127
#23 trace_call_function (kwnames=<optimized out>, nargs=<optimized out>, args=<optimized out>, func=<optimized out>, tstate=<optimized out>) at Python/ceval.c:4968
#24 call_function (kwnames=0x0, oparg=<optimized out>, pp_stack=<synthetic pointer>, tstate=0x5770c80) at Python/ceval.c:4984
#25 _PyEval_EvalFrameDefault (f=<optimized out>, throwflag=<optimized out>) at Python/ceval.c:3486
#26 0x00007f1e2d04eca0 in function_code_fastcall (co=<optimized out>, args=<optimized out>, nargs=2, globals=<optimized out>) at Objects/call.c:283
#27 0x00007f1e2d050c4b in _PyObject_Vectorcall (kwnames=<optimized out>, nargsf=<optimized out>, args=<optimized out>, callable=<optimized out>)
    at ./Include/cpython/abstract.h:127
#28 trace_call_function (kwnames=<optimized out>, nargs=<optimized out>, args=<optimized out>, func=<optimized out>, tstate=<optimized out>) at Python/ceval.c:4968
#29 call_function (kwnames=0x0, oparg=<optimized out>, pp_stack=<synthetic pointer>, tstate=0x5770c80) at Python/ceval.c:4984
#30 _PyEval_EvalFrameDefault (f=<optimized out>, throwflag=<optimized out>) at Python/ceval.c:3486
#31 0x00007f1e2d1726ee in _PyEval_EvalCodeWithName (_co=0x7f1e2b7112f0, globals=<optimized out>, locals=locals@entry=0x0, args=<optimized out>, argcount=2,
    kwnames=0x7f1e2b773568, kwargs=kwargs@entry=0x7f1e2b586400, kwcount=1, kwstep=kwstep@entry=1, defs=defs@entry=0x7f1e2b767778, defcount=defcount@entry=1,
    kwdefs=kwdefs@entry=0x0, closure=closure@entry=0x0, name=0x7f1e2b768130, qualname=0x7f1e2b70f670) at Python/ceval.c:4298
#32 0x00007f1e2d07ecb2 in _PyFunction_Vectorcall (func=<optimized out>, stack=<optimized out>, nargsf=<optimized out>, kwnames=<optimized out>) at Objects/call.c:435
#33 0x00007f1e2d082b31 in _PyObject_Vectorcall (kwnames=0x7f1e2b773550, nargsf=2, args=0x7f1e2b5863f0, callable=0x7f1e2b715d30) at ./Include/cpython/abstract.h:127
#34 method_vectorcall (method=<optimized out>, args=0x7f1e2b5863f8, nargsf=<optimized out>, kwnames=0x7f1e2b773550) at Objects/classobject.c:60
#35 0x00007f1e2d05661b in _PyObject_Vectorcall (kwnames=<optimized out>, nargsf=<optimized out>, args=<optimized out>, callable=<optimized out>)
    at ./Include/cpython/abstract.h:127
#36 trace_call_function (kwnames=<optimized out>, nargs=<optimized out>, args=<optimized out>, func=<optimized out>, tstate=<optimized out>) at Python/ceval.c:4968
#37 call_function (kwnames=0x7f1e2b773550, oparg=<optimized out>, pp_stack=<synthetic pointer>, tstate=0x5770c80) at Python/ceval.c:4984
[... cut ...]
#49 0x00007f1e33166bff in GpiCommonCbHdl::run_callback (this=0x576ab38) at cocotb/share/lib/gpi/GpiCbHdl.cpp:86
#50 0x00007f1e333884c4 in handle_vpi_callback (cb_data=0x7f1e36019b10) at cocotb/share/lib/vpi/VpiImpl.cpp:652
#51 0x0000000000706c2e in ?? ()
#52 0x0000000000adee82 in ?? ()
#53 0x0000000004c2503a in ?? ()
#54 0x00007f1e36019bf8 in ?? ()
#55 0x0000000005bb1801 in ?? ()
#56 0xffffffffffffffff in ?? ()
#57 0x00000000057a0450 in ?? ()
#58 0x0000000000000000 in ?? ()
+cont
Continuing.
[Inferior 1 (process 31026) exited with code 01]
+quit
```

Toucanfan, 2023-01-18:
The main thing I spot is, that it is the same signal that is set in each call to gpi_set_signal_value_int. The first time it is set to 0, whereas the next time it is set to 1.

Toucanfan, 2023-01-18:
As for the following:
```
#1  0x000000000082f364 in ?? ()
#2  0x0000000000c2a5a9 in ?? ()
#3  0x0000000000b6c4d8 in ?? ()
#4  0x0000000000b58d88 in ?? ()
#5  0x000000000089a84f in ?? ()
```

it looks like these are functions in the simulator binary, `xmsim`, since these addresses belong to a memory mapped region of
the binary.

marlonjames, 2023-01-18:
Yeah, it's what I expected.

These are the key parts:
```
#0  handle_vhpi_callback (cb_data=0x28d6ae0) at cocotb/share/lib/vhpi/VhpiImpl.cpp:988
[ ... simulator ... ]
#6  0x00000000008992ca in vhpi_put_value ()
#7  0x00007f1e2d388710 in VhpiLogicSignalObjHdl::set_signal_value (this=0x5d68cc0, value=1, action=GPI_DEPOSIT) at cocotb/share/lib/vhpi/VhpiCbHdl.cpp:527
#8  0x00007f1e33164ad9 in gpi_set_signal_value_int (sig_hdl=0x5d68cc0, value=1, action=GPI_DEPOSIT) at cocotb/share/lib/gpi/GpiCommon.cpp:483
#9  0x00007f1e2c01cad6 in set_signal_val_int (self=0x7f1e2b0a2d70, args=0x7f1e2a923e00) at cocotb/share/lib/simulator/simulatormodule.cpp:657
[ ... Python (cocotb) ... ]
#48 0x00007f1e2c01bd8b in handle_gpi_callback (user_data=0x5d7bc00) at cocotb/share/lib/simulator/simulatormodule.cpp:228
#49 0x00007f1e33166bff in GpiCommonCbHdl::run_callback (this=0x576ab38) at cocotb/share/lib/gpi/GpiCbHdl.cpp:86
#50 0x00007f1e333884c4 in handle_vpi_callback (cb_data=0x7f1e36019b10) at cocotb/share/lib/vpi/VpiImpl.cpp:652
```

1. While cocotb is running in response to a callback, it calls down through the `simulator` module to set a value, which goes to the GPI and then back to the simulator with `vhpi_put_value()`.
2. The simulator updates the value, checks for registered callbacks, then calls the callback immediately.

This is non-compliant with the VHDL LRM. The details are here: https://github.com/cocotb/cocotb/issues/2178#issuecomment-722789720

Toucanfan, 2023-01-19:
Okay, but why does it work the first time then? Also note, that frame #50 is handle_vpi_callback, not handle_vhpi_callback. And this is the case both times the breakpoint is hit.

marlonjames, 2023-01-19:
You are right, I just noticed **VPI** then **VHPI**. Mixed language simulation might be part of the problem here.

Toucanfan, 2023-01-19:
Yes, I do have some Verilog modules deep down in the hierachy, but not any that I interact with from my testbench. Anyway, thank you for the debug effort.

marlonjames, 2023-01-19:
If you step through the first breakpoint that "works", does it actually call up to Python, or is perhaps the trigger not primed?

Toucanfan, 2023-01-19:
From what I can deduce, it look like the callback is not really called, just cleaned up immediately.

```
+b handle_vhpi_callback
Breakpoint 1 at 0x7f0b6fe3aa6d: file cocotb/share/lib/vhpi/VhpiImpl.cpp, line 988.
+cont
Continuing.
warning: Corrupted shared library list: 0x65b1370 != 0x659d710

Breakpoint 1, handle_vhpi_callback (cb_data=0x28d6ae0) at cocotb/share/lib/vhpi/VhpiImpl.cpp:988
988         gpi_to_user();
+next
990         VhpiCbHdl *cb_hdl = (VhpiCbHdl *)cb_data->user_data;
+next
992         if (!cb_hdl) {
+next
998         gpi_cb_state_e old_state = cb_hdl->get_call_state();
+next
1000        if (old_state == GPI_PRIMED) {
+next
1001            cb_hdl->set_call_state(GPI_CALL);
+next
1002            cb_hdl->run_callback();
+step
GpiValueCbHdl::run_callback (this=0x65bb388) at cocotb/share/lib/gpi/GpiCbHdl.cpp:111
111         std::string current_value;
+next
114         if (required_value == "X")
+next
117             current_value = m_signal->get_signal_value_binstr();
+next
118             if (current_value == required_value) pass = true;
+next
118             if (current_value == required_value) pass = true;
+next
112         bool pass = false;
+next
121         if (pass) {
+next
124             cleanup_callback();
+next
125             arm_callback();
+next
128         return 0;
+next
129     }
+next
handle_vhpi_callback (cb_data=0x28d6ae0) at cocotb/share/lib/vhpi/VhpiImpl.cpp:1004
1004            gpi_cb_state_e new_state = cb_hdl->get_call_state();
+next
1007            if (new_state != GPI_PRIMED)
+next
1013        gpi_to_simulator();
+next
1014    };
+next
0x000000000082f364 in ?? ()
+quit
```

marlonjames, 2023-01-19:
It looks like you have either a `RisingEdge` or `FallingEdge` Trigger being `await`-ed by a cocotb Task, but for this callback, the value doesn't match so it never propogates up to cocotb.

The next time, the value has changed to match the Trigger, so it does make it up to cocotb.

Toucanfan, 2023-01-19:
Okay, I see. Yes, I do have a RisingEdge trigger, and it should fire around 50ns into the simulation, so it makes sense.

commit 88151f27 (2024-01-02): Xcelium: Set feature flag to make VHDL toplevels work

Xcelium 23.09.004 (and some other hotfix releases) contain a fix for
https://github.com/cocotb/cocotb/issues/1076, which is enabled when
setting a specific feature flag. Always set this flag when VHDL
toplevels are used in cocotb.

Fixes #1076
commit 7f4ac481 (2024-06-19): Xcelium: Set feature flag to make VHDL toplevels work

Xcelium 23.09.004 (and some other hotfix releases) contain a fix for
https://github.com/cocotb/cocotb/issues/1076, which is enabled when
setting a specific feature flag. Always set this flag when VHDL
toplevels are used in cocotb.

Fixes #1076

(cherry picked from commit 88151f270f3eeda9144bae9ae39d0018887ae8f2)
closed 2024-01-02 by imphil